Universal-Debloater-Alliance/universal-android-debloater-next-generation #546: pkg(scope): com.coloros.colordirectservice

aryanpednekar45, 2024-07-10:
### List your packages

- Package name: Screen Recognition
- Current list: Recommended
- Proposed list: Recommended
- Current removal: 
- Proposed removal: 
- Current description: Boo awareness screen
Smart things, also have something to Breeno touch
- Proposed description: Screen Recognition (Breeno Touch)
This app is used to recognize screen contents and provide quick access to relevant services (like Text Extraction, AI Summarizer)
(com.coloros.ocrscanner, com.oplus.ocs, com.oplus.aiunit, com.coloros.ocs.opencapabilityservice are also needed for this to work)

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-07-14:
I'm making it `Advanced`.

aryanpednekar45, 2024-07-15:
> I'm making it `Advanced`.

Recommended would be better. Uninstalling it doesn't have any issues.

AnonymousWP, 2024-07-15:
The Wiki is leading here: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ

> Recommended -- Pointless or outright negative packages, and/or apps available through Google Play.
> Advanced -- Breaks obscure or minor parts of functionality, or apps that aren't easily enabled/installed through Settings/Google Play. This category is also used for apps that are useful (default keyboard/gallery/launcher/music app.) but that can easily be replaced by a better alternative.

aryanpednekar45, 2024-07-16:
> The Wiki is leading here: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ

Recommended. It's not a useful app and uninstall it doesn't do any harm

aryanpednekar45, 2024-07-16:
> I'm making it `Advanced`.

Ok 👍🏻

closed 2024-09-02 by AnonymousWP