Universal-Debloater-Alliance/universal-android-debloater-next-generation #550: bug(theme): `Auto` partially updates views, when changing DE color-scheme

Rudxain, 2024-07-10:
### Describe the bug

https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/531#issuecomment-2220897348

New repro:
1. Go to Settings
2. Ensure Auto theme is set
3. Change desktop theme
4. Click any empty space in the Settings View

### Expected behavior

Either:
- view shouldn't update colors until switching back-&-forth between views
- full view gets an updated palette

### Provide logs

> (to-do)

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Frigyes06, 2024-07-10:
Can reproduce on Arch linux

AnonymousWP, 2024-08-06:
@Universal-Debloater-Alliance/developers Can we fix this so we can release 1.1.0?

Rudxain, 2024-08-06:
I might attempt to fix it, but I have to read the Iced docs first

AnonymousWP, 2024-09-07:
Removed this from the v1.1.0 release milestone as it's likely something we can't fix.

lavafroth, 2024-09-12:
The theme will change if you restart the program. Sadly, this is more of an iced-rs issue.

closed 2024-11-03 by adhirajsinghchauhan