Universal-Debloater-Alliance/universal-android-debloater-next-generation #553: Some needed application has been deleted after

darkseidH, 2024-07-12:
### Describe the problem

I have a nothing phone when i do debloating the number of packages in the recomended is so big and it removes essentiel things lik fingerprint weather app 

### Provide logs

[Uploading UAD_20240712.log…]()

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-07-14:
You didn't upload your logs properly. Tell us which app is causing issues and we'll take proper action.

schbrns, 2024-11-24:
@darkseidH Could you please follow-up on this issue? Thank you.

AnonymousWP, 2025-01-06:
Closing due to inactivity. This can be re-opened once the original poster posted logs.

closed 2025-01-06 by AnonymousWP