Universal-Debloater-Alliance/universal-android-debloater-next-generation #556: bug(exec): Can't open window

Rudxain, 2024-07-15:
### Describe the bug

UAD opens the window briefly, then it closes itself. Possibly a crash/panic

### Expected behavior

It must open (obviously)

### You have a solution?

> [I suspect Microsoft or NVIDIA pushed an update that broke UAD](https://discord.com/channels/1168607797081022514/1168607798196715573/1262273395316232253)

### Operating System

Windows

### Provide logs

_None 💀_

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Alpha-Radke, 2024-07-15:
Yes I can confirm this is 99% because of older GPU drivers that I started using for my GTX 1080 on fresh Windows 10 (22H2 19045.3448)
NVIDIA Driver 385.69
No iGPU Intel i3-10105F
[UAD_20240712.log](https://github.com/user-attachments/files/16233341/UAD_20240712.log)

Frigyes06, 2024-07-25:
Cannot repro on Win 10 and an AMD video card

AnonymousWP, 2024-09-15:
If it's related to GPU drivers, then is it even something we can fix?

Rudxain, 2024-09-16:
> something we can fix?

I doubt it. Even if it was a problem in the program, it would most likely be related to `wgpu`.

Maybe the bug has been fixed since dependencies were updated?

Anyways, I'm closing this because of lack of info

closed 2024-09-16 by Rudxain