Universal-Debloater-Alliance/universal-android-debloater-next-generation #557: pkg(scope): com.miui.guardprovider Change description)

null-von-sushi, 2024-07-18:
### List your packages

- Package name: com.miui.guardprovider
- Current list: oem
  - Proposed list: oem
- Current removal: advanced
  - Proposed removal: advanced
- Current description: 
```
Guard Provider security app\nThe app includes 3 different antivirus brands built in that the user can choose (Avast, AVL and Tencent). \nThis app notably performs a virus scan of any apps you want to install. \nA serious vulnerability was found in 2019\nWorth reading : https://research.checkpoint.com/2019/vulnerability-in-xiaomi-pre-installed-security-app/\nYou may want to remove this app from a privacy stance.\nhttps://beta.pithus.org/report/797a7e405bc8e767deebbbcab3e06a19b05156de44292c918b582dff3078d7b8\nIMPORTANT NOTE: Removing this package will very likely break any app installation/update.\nBefore removing, disable security things in MIUI security app.
```
  - Proposed description: 
  ```
Guard Provider security app\nThe app includes 3 different antivirus brands built in that the user can choose (Avast, AVL and Tencent). \nThis app notably performs a virus scan of any apps you want to install. \nA serious vulnerability was found in 2019\nWorth reading : https://research.checkpoint.com/2019/vulnerability-in-xiaomi-pre-installed-security-app/\nYou may want to remove this app from a privacy stance.\nhttps://beta.pithus.org/report/797a7e405bc8e767deebbbcab3e06a19b05156de44292c918b582dff3078d7b8\nIMPORTANT NOTE: Removing this package will very likely break any app installation/update.\nBefore removing, disable security things in MIUI security app.\nOn HyperOS this module is responsible for showing the 'Install via USB' prompt when installing apps via ADB (enabled through developer options, requires Mi account the first time). 
  ```

In short, I just wanted to mention that this is specifically responsible for the confirmation prompt when using ADB install or something like 'Install with Options' (necessary to install apps that target a low SDK version for example).
You need to add a mi account to enable that first time, but afterwards you can disable the mi account components it seems and it seems to keep working, though disabling `com.miui.guardprovider` itself will cause any install to fail.

I am not sure what the paragraph talking mentioning that  'removing this package will very likely break any app installation/update' is talking about, as the phrasing implies that normal installs via either Google Play or APK files (via Package Installer) will fail, however that seems to not be the case? Since it may be relevant to other devices or older MIUI versions I just kept it.

I hope this is okay.

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-09-18:
Hi, sorry for the very late action on this. Could you tell me if this is okay? See linked pull request.

null-von-sushi, 2024-09-18:
Yup, seems fine to me, thank you!

closed 2024-09-25 by adhirajsinghchauhan