Universal-Debloater-Alliance/universal-android-debloater-next-generation #558: pkg(scope): com.samsung.adnroid.knox.app.networkfilter uninstalling crashes Bluetooth

Tirgatt, 2024-07-19:
### Describe the problem

After uninstalling com.samsung.android.knox.app.networkfilter, Bluetooth on my Samsung Galaxy Z Flip 6 keeps crashing with the message 'Bluetooth keeps stopping'.

Device Specifications (as shipped):
One UI Version: 6.1.1
Android Version: 14
Security Patch Level: June 1, 2024
Baseband Version: F741BXXU1AXFD

Additional Information:
I purchased the device directly from the Samsung Shop App.
The package did not come directly from Samsung; the label read "Kuehne & Nagel (AG & Co.) KG".

### You have a solution?

If I restore com.samsung.android.knox.app.networkfilter, Bluetooth works fine again.

### Provide logs

[UAD_20240719.log](https://github.com/user-attachments/files/16316440/UAD_20240719.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Tirgatt, 2024-07-26:
I forgot to say that I used the "Recommended" filter when I uninstalled com.samsung.android.knox.app.networkfilter. I did not expect a package marked as "Recommended" to cause essential functions like Bluetooth to stop working.

king-gheedorah, 2024-07-27:
Experienced the same issue and solution.

AnonymousWP, 2024-09-18:
Thanks for telling us. I'll fix this.

king-gheedorah, 2024-09-18:
> Does this look good to you both? @Tirgatt[https://github.com/Tirgatt] @king-gheedorah[https://github.com/king-gheedorah]

Looks good, thanks!

closed 2024-09-25 by adhirajsinghchauhan