Universal-Debloater-Alliance/universal-android-debloater-next-generation #559: feat(scope): Warn on app Uninstall if Restore is not possible

RobbieGordon, 2024-07-21:
### Describe the feature you want

I am unable to restore apps on a Motorola Moto G Play (Android 7.1.1).  "cmd package install-existing" is not recognized.
[UAD_20240720_motorola.log](https://github.com/user-attachments/files/16323684/UAD_20240720_motorola.log)

Possibly related to Android 7.

If this is a known issue, UAD-NG should give a warning before uninstalling apps where the restore will not work.

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-07-29:
Please re-open an issue using the bug report template instead.

RobbieGordon, 2024-07-31:
> DuckDuckGo removed one tracker. More<https://duckduckgo.com/-madpzLWtiyLxsQiLNpApVFgBRYgpm1NoYKgoI9X5rad_dIaIBtoMcKTheRgWKmZyaY2odsCQpwlKe0rkUbgA-S4iTBwl3pVzaTSC9N6VI8BLExpcYIa9Io10G8HiAmz2J0-sPxs3dzvD0zoe-ep_uleCVjcrivsaDDa7Q13MdPN_fQStd3TR-X1JfUeLdT7accBYG1NjaBeXSRdTvvd--wHvvtle_vBu9v5e8oAb2trD-TyYml3Loc3RTk3YKCOJAPh-R0LyIY2sMONMB-54rRrKJsSHffgCYwH9IiSUPLsgJ2YkvKR4ZORMtO0wQwX0rSjkKWsTaWhqhJZiS3pKSAh_wWumF3sv8zoLm4-SFwWEEjy3qFdZlznYxLKY7tN25rT3W9V2fO8ZUlUBfEz0pYau-0oltio1MkGEPcDX6PzhRTtGjaXMEPE-wO9FZoLjB2_GAtIf_8KkhSW5jvG9zmYtnJafFXiKqQQiWiDDg>
> Deactivate <https://duckduckgo.com/>

This issue is not really a bug with UAD-ng.
Using ADB on my Android 7 phone, 'cmd package install-existing' does not work, whereas 'pm install' does.
As a result, using UAD-ng on this phone, I am able to uninstall packages but not restore them.

AnonymousWP, 2024-08-14:
But then it should be a native feature of UAD-ng. I'll re-open it and change the title.