Universal-Debloater-Alliance/universal-android-debloater-next-generation #562: bug(scope): Crash on Intel HD Graphics 620 (i5-7200u) graphics card on Windows 10 1607

yjulm, 2024-07-25:
### Describe the bug

Version 1.0.2 crashed before the interface was displayed. I tried upgrading to version 1.0.3 and it still crashed.

### Expected behavior

Since many software no longer support Windows 8.1, I recently upgraded to Windows 10. Everything was working fine with Windows 8.1.

### Operating System

Windows

### Provide logs

2024-07-25 13:01:42 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.19.3\src\vulkan\instance.rs:291] Unable to find extension: VK_EXT_swapchain_colorspace
2024-07-25 13:01:42 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.19.3\src\auxil\dxgi\factory.rs:214] Failed to cast IDXGIFactory4 to IDXGIFactory6: "0x80004002"
2024-07-25 13:01:42 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.19.3\src\dx12\adapter.rs:63] Device creation failed: 0x887A0004
2024-07-25 13:01:42 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.19.3\src\dx12\adapter.rs:63] Device creation failed: 0x887A0004
2024-07-25 13:01:42 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-core-0.19.3\src\instance.rs:342] Missing downlevel flags: DownlevelFlags(SURFACE_VIEW_FORMATS)
The underlying API or device in use does not support enough features to be a fully compliant implementation of WebGPU. A subset of the features can still be used. If you are running this program on native and not in a browser and wish to limit the features you use to the supported subset, call Adapter::downlevel_properties or Device::downlevel_properties to get a listing of the features the current platform supports.
2024-07-25 13:01:42 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-core-0.19.3\src\instance.rs:347] DownlevelCapabilities {
    flags: DownlevelFlags(
        COMPUTE_SHADERS | FRAGMENT_WRITABLE_STORAGE | INDIRECT_EXECUTION | BASE_VERTEX | READ_ONLY_DEPTH_STENCIL | NON_POWER_OF_TWO_MIPMAPPED_TEXTURES | CUBE_ARRAY_TEXTURES | COMPARISON_SAMPLERS | INDEPENDENT_BLEND | VERTEX_STORAGE | ANISOTROPIC_FILTERING | FRAGMENT_STORAGE | MULTISAMPLED_SHADING | DEPTH_TEXTURE_AND_BUFFER_COPIES | WEBGPU_TEXTURE_FORMAT_SUPPORT | BUFFER_BINDINGS_NOT_16_BYTE_ALIGNED | UNRESTRICTED_INDEX_BUFFER | FULL_DRAW_INDEX_UINT32 | DEPTH_BIAS_CLAMP | VIEW_FORMATS | UNRESTRICTED_EXTERNAL_TEXTURE_COPIES | NONBLOCKING_QUERY_RESOLVE | VERTEX_AND_INSTANCE_INDEX_RESPECTS_RESPECTIVE_FIRST_VALUE_IN_INDIRECT_DRAW,
    ),
    limits: DownlevelLimits,
    shader_model: Sm5,
}

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Marekk2k, 2024-07-25:
have you tried downloading version 1.0.3 directly from github rather than by updating?

AnonymousWP, 2024-08-10:
Hi, see https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/36 please.

AnonymousWP, 2024-09-15:
Closing this as a solution was suggested. Also, please try v1.1.0.

closed 2024-09-15 by AnonymousWP