Universal-Debloater-Alliance/universal-android-debloater-next-generation #568: bug(scope): Backup restore does not apply to secondary users

ticklemyIP, 2024-07-28:
### Describe the bug

When trying to restore a backup .json it does not get applied to the secondary user. Even though in the .json there is a separate list for every user, the backup function does not work (in my case the secondary user 11) and leaves behind some if not all the packages marked as Uninstalled in the .json file.

For example in the backup file both user (0 and 11) has the youtube package marked as removed because it was removed in both profiles when the backup file was created. When restoring the backup file youtube is only removed in the primary (0 profile) and is still present in the secondary (11) profile.

### Expected behavior

The backup function applies the backup to secondary users.

### You have a solution?

As you can see from the logs, the secondary user is nowhere to be seen, I tried making a TEST.json file that only contains the secondary users packages, which still did not work. See the logs below.

Device model can be seen in the logs.

Either fix this current issue, or bring back the old export .txt function. It was reliable and there is no reason for both of the to exist at the same time. Especially because to the best of my knowledge there is currently no way to remove packages per user, or mass select them from list (like the old UAD program).

### Operating System

Linux

### Window Manager

Wayland

### Desktop Environment

GNOME

### Provide logs

2024-07-28 17:25:04 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-07-28 17:25:04 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-A528B
2024-07-28 17:25:04 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------
2024-07-28 17:25:09 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-07-28 17:25:09 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-A528B
2024-07-28 17:25:09 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------
2024-07-28 17:25:10 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-07-28 17:25:10 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-A528B
2024-07-28 17:25:10 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------
2024-07-28 17:25:10 WARN  [src/core/uad_lists.rs:208] Could not load remote debloat list
2024-07-28 17:25:17 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-07-28 17:25:18 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-07-28 17:25:18 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-A528B
2024-07-28 17:25:18 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------
2024-07-28 17:25:19 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-07-28 17:25:19 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-A528B
2024-07-28 17:25:19 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------
2024-07-28 17:25:19 WARN  [src/core/uad_lists.rs:208] Could not load remote debloat list
2024-07-28 17:25:24 INFO  [src/gui/views/settings.rs:165] [RESTORE] Restoring backup 2024-07-27_20-45-14
2024-07-28 17:25:24 INFO  [src/core/sync.rs:123] [RESTORE] pm uninstall --user 0 com.google.android.googlequicksearchbox -> Success
2024-07-28 17:25:24 INFO  [src/core/sync.rs:123] [RESTORE] pm uninstall --user 0 com.google.android.projection.gearhead -> Success
2024-07-28 17:25:24 INFO  [src/core/sync.rs:123] [RESTORE] pm uninstall --user 0 com.sec.android.app.vepreload -> Success
2024-07-28 17:25:26 INFO  [src/core/sync.rs:123] [RESTORE] cmd package install-existing --user 0 com.google.android.googlequicksearchbox -> Package com.google.android.googlequicksearchbox installed for user: 0
2024-07-28 17:25:26 INFO  [src/core/sync.rs:123] [RESTORE] cmd package install-existing --user 0 com.google.android.projection.gearhead -> Package com.google.android.projection.gearhead installed for user: 0
2024-07-28 17:25:27 INFO  [src/core/sync.rs:123] [RESTORE] cmd package install-existing --user 0 com.sec.android.app.vepreload -> Package com.sec.android.app.vepreload installed for user: 0
2024-07-28 17:25:27 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-07-28 17:25:27 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-A528B
2024-07-28 17:25:27 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------
2024-07-28 17:25:28 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-07-28 17:25:28 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-A528B
2024-07-28 17:25:28 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------
2024-07-28 17:25:28 WARN  [src/core/uad_lists.rs:208] Could not load remote debloat list
2024-07-28 17:25:35 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-07-28 17:25:35 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-07-28 17:25:35 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-A528B
2024-07-28 17:25:35 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------
2024-07-28 17:25:36 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-07-28 17:25:36 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-A528B
[... 87 lines cut ...]

-----TEST BACKUP LOGS-----

2024-07-28 17:51:23 ERROR [src/core/sync.rs:129] [Shell] reboot -> 
2024-07-28 17:51:26 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-07-28 17:52:05 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-07-28 17:52:05 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-A528B
2024-07-28 17:52:05 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------
2024-07-28 17:52:28 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-07-28 17:52:35 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-07-28 17:52:35 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-A528B
2024-07-28 17:52:35 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------
2024-07-28 17:52:52 INFO  [src/gui/views/settings.rs:165] [RESTORE] Restoring backup TEST
2024-07-28 17:52:53 INFO  [src/core/sync.rs:123] [RESTORE] cmd package install-existing --user 0 com.google.android.googlequicksearchbox -> Package com.google.android.googlequicksearchbox installed for user: 0
2024-07-28 17:52:53 INFO  [src/core/sync.rs:123] [RESTORE] cmd package install-existing --user 0 com.google.android.projection.gearhead -> Package com.google.android.projection.gearhead installed for user: 0
2024-07-28 17:52:54 INFO  [src/core/sync.rs:123] [RESTORE] cmd package install-existing --user 0 com.sec.android.app.vepreload -> Package com.sec.android.app.vepreload installed for user: 0
2024-07-28 17:52:55 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-07-28 17:52:55 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-A528B
2024-07-28 17:52:55 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-08-22:
Hi, sorry for the late reply. Have you found the cause already? And, more importantly, could you help out by creating a PR?

ticklemyIP, 2024-08-22:
I don't really know very basic code so I couldn't do troubleshooting if you put a gun to my head, so unless you want me to help out writing a wiki or something I can't provide a PR. Glad to hear the old export's coming back. I can maBe help with testing though. I'm curious am I the only one with this bug or is it universal?

AnonymousWP, 2024-08-23:
Wiki is not needed indeed, this is purely Rust-related.

I think you're the only one so far, but it's certainly a feature request rather than a bug.

AnonymousWP, 2024-09-15:
v1.1.0 should solve your problem. Closing this now. Let us know if you still face issues.

closed 2024-09-15 by AnonymousWP