Universal-Debloater-Alliance/universal-android-debloater-next-generation #570: feat(scope): Set "Affect all users of the device" setting to off by default or ask user

ticklemyIP, 2024-07-28:
### Describe the feature you want

There is an "Affect all users of the device" setting.
I did not notice it when modifying packages because I never knew this feature was implemented; and 2: i cannot imagine a feature like this would exist and would be turned on by default.

When the app first starts:
Detect if the user has multiple profiles. If yes, ask the user if they want to have this setting turned on. Save their preference for future setting.

Alternatively (or simultaniously) it would probably be better for this setting to be **turned off by default** when the app starts.
Moving this setting to a more visible place and **greying out the user selection dropdown** when this is enabled to indicate it's presence would also be useful.

I expected the app to remove only packages per user profile, because as a veteran user that functionality is what I was used to and any reasonable person would expect the app to work per user if the ability to select a user from a dropdown is present (and not greyed out).

As a result my work profile and main profile is messed up because packages with the same name that I wanted to remove in the main profile I may not have been wanting to remove in my work profile and vice versa.

For example there are apps like OTA update packages which by default are not present in the work profile because they are not needed. If I was careless I could have enabled that package in the work profile unneeded or what's worse, could have removed it in the main profile when it would be very much needed.

Please consider changing this feature.

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

starbrights, 2025-02-12:
I very much miss the option to handle debloating  separate for each user (usually 0 and 10 for the work profile) - or at least have an option.
You can already select the user. But there is a message: The action is applied for ALL users. 
That is most likely not what you want. 
Can you please add this feature?

AnonymousWP, 2025-03-08:
@starbrights Could you file a pull request for that? We don't have many active developers to add many new features unfortunately.