Universal-Debloater-Alliance/universal-android-debloater-next-generation #571: Recommended apps breaks System Navigation

akarn911, 2024-07-28:
### Describe the problem

These five recommended apps breaks the system navigation when switching between 3-button and gestures.
( I am using Motorola phone - Moto E13 )
1. com.google.android.overlay.gmsconfig.comms
2.com.google.android.overlay.gmsconfig.geotz
3. com.google.android.overlay.gmsconfig.personalsafety
4. com.google.android.overlay.modules.modulemetadata.forframework
5. com.google.android.overlay.modules.permissioncontroller.forframework

### You have a solution?

Remove them from recommended list.

### Provide logs

I don't know what is this

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

schbrns, 2024-09-13:
I mean, you shouldn't be removing overlay modules. Are these not set to expert?

akarn911, 2024-09-13:
These are in recommended list.

AnonymousWP, 2024-10-05:
You don't need to close your issue:

> [AnonymousWP](https://github.com/AnonymousWP) linked a pull request [33 minutes ago](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/571#event-14527466009) that will close this issue

akarn911, 2024-10-05:
Ok.
Thank you 👍

closed 2024-10-20 by Rudxain