Universal-Debloater-Alliance/universal-android-debloater-next-generation #574: bug(scope): unable to uninstall com.oplus.cosa

huskyyyyy, 2024-08-04:
### Describe the bug

Happened on the original UAD and latest next gen version . Please fix it. Thank you.

### Expected behavior

It should uninstall the bloatware when I press uninstall, but nothing happens when I press it. 

### Operating System

Windows

### Provide logs

N/A

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-08-05:
Try disabling instead, that should work.

closed 2024-08-10 by AnonymousWP

huskyyyyy, 2024-08-19:
I don't see the disable button in the ad console. And there's nowhere to disable in my OnePlus phone as well. Can you elaborate more on it? Thanks

AnonymousWP, 2024-08-19:
Go to the settings of UAD-ng. There you should see a checkbox regarding disabling.