Universal-Debloater-Alliance/universal-android-debloater-next-generation #580: bug(scope):

rafnov, 2024-08-07:
### Describe the bug

Universal-android-debloater does not start with errors.
I'm on Archlinux and we have universal-android-debloater 1.0.3-1 in AUR.
https://aur.archlinux.org/packages?O=0&SeB=nd&K=universal+android&outdated=&SB=p&SO=d&PP=50&submit=Go
After very long compiling it does not even start with errors that I paste as RUST_BACKTRACE.
uad-ng-bin neither starts with another RUST errors.
Only one I was able to run was universal-android-debloater-opengl-bin, which is very outdated there.
How to fix?

### Expected behavior

Universal-android-debloater running.

### Operating System

Linux

### Window Manager

Xorg

### Desktop Environment

Xfce

### Provide logs

[rav: ~]$ RUST_BACKTRACE=full uad-ng
2024-08-07 09:04:35 WARN  [/tmp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/instance.rs:291] Unable to find ex
tension: VK_KHR_surface
2024-08-07 09:04:35 WARN  [/tmp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/instance.rs:291] Unable to find ex
tension: VK_KHR_xlib_surface
2024-08-07 09:04:35 WARN  [/tmp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/instance.rs:291] Unable to find ex
tension: VK_KHR_xcb_surface
2024-08-07 09:04:35 WARN  [/tmp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/instance.rs:291] Unable to find ex
tension: VK_KHR_wayland_surface
2024-08-07 09:04:35 WARN  [/tmp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/instance.rs:291] Unable to find ex
tension: VK_EXT_swapchain_colorspace
2024-08-07 09:04:35 WARN  [/tmp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/instance.rs:291] Unable to find ex
tension: VK_KHR_get_physical_device_properties2
thread 'main' panicked at /tmp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-core-0.19.3/src/instance.rs:521:39:
called `Option::unwrap()` on a `None` value
stack backtrace:
   0:     0x559b6548e78f - <unknown>
   1:     0x559b65162e5b - <unknown>
   2:     0x559b6545f0ae - <unknown>
   3:     0x559b65493629 - <unknown>
   4:     0x559b65492f1b - <unknown>
   5:     0x559b65494183 - <unknown>
   6:     0x559b6549396a - <unknown>
   7:     0x559b654938f9 - <unknown>
   8:     0x559b654938e6 - <unknown>
   9:     0x559b650c1b52 - <unknown>
  10:     0x559b650c1ceb - <unknown>
  11:     0x559b650c20e8 - <unknown>
  12:     0x559b65700c85 - <unknown>
  13:     0x559b650ffced - <unknown>
  14:     0x559b650f8858 - <unknown>
  15:     0x559b655dedf7 - <unknown>
  16:     0x559b655ae45e - <unknown>
  17:     0x559b65626abf - <unknown>
  18:     0x559b655ad65d - <unknown>
  19:     0x7699654e2e08 - <unknown>
  20:     0x7699654e2ecc - __libc_start_main
  21:     0x559b650f4a25 - <unknown>
  22:                0x0 - <unknown>

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

rafnov, 2024-08-07:
This is trying to run from uad-ng-bin package:
```
[rav: ~]$ RUST_BACKTRACE=full uad-ng
2024-08-07 09:17:40 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/instance.rs:291] Unable to
 find extension: VK_KHR_surface
2024-08-07 09:17:40 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/instance.rs:291] Unable to
 find extension: VK_KHR_xlib_surface
2024-08-07 09:17:40 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/instance.rs:291] Unable to
 find extension: VK_KHR_xcb_surface
2024-08-07 09:17:40 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/instance.rs:291] Unable to
 find extension: VK_KHR_wayland_surface
2024-08-07 09:17:40 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/instance.rs:291] Unable to
 find extension: VK_EXT_swapchain_colorspace
2024-08-07 09:17:40 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/instance.rs:291] Unable to
 find extension: VK_KHR_get_physical_device_properties2
thread 'main' panicked at /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-core-0.19.3/src/instance.rs:521:39:
called `Option::unwrap()` on a `None` value
stack backtrace:
   0:     0x5a541faff208 - <unknown>
   1:     0x5a541f7954e0 - <unknown>
   2:     0x5a541fac957e - <unknown>
   3:     0x5a541fb00cde - <unknown>
   4:     0x5a541fb00605 - <unknown>
   5:     0x5a541fb01517 - <unknown>
   6:     0x5a541fb01008 - <unknown>
   7:     0x5a541fb00f96 - <unknown>
   8:     0x5a541fb00f83 - <unknown>
   9:     0x5a541f793634 - <unknown>
  10:     0x5a541f795132 - <unknown>
  11:     0x5a541f79cb75 - <unknown>
  12:     0x5a541fd7085b - <unknown>
  13:     0x5a541fc2370c - <unknown>
  14:     0x5a541fc88015 - <unknown>
  15:     0x5a541fc4e072 - <unknown>
  16:     0x5a541f727316 - <unknown>
  17:     0x5a541fc61f02 - <unknown>
  18:     0x5a541fc7510e - <unknown>
  19:     0x5a541fc61054 - <unknown>
  20:     0x727dc3db1e08 - <unknown>
  21:     0x727dc3db1ecc - __libc_start_main
  22:     0x5a541f71dbe5 - <unknown>
  23:                0x0 - <unknown>
```
https://aur.archlinux.org/packages/uad-ng-bin

AnonymousWP, 2024-08-07:
Please edit your title. It's sort of empty.

rafnov, 2024-08-09:
Sorry, I'm new here and posted it from basic Android app. OK now?

AnonymousWP, 2024-08-10:
I don't know, maybe you're clicking wrong buttons; I'm not closing it.

Anyway, see https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/36. Try that please.

AnonymousWP, 2024-09-15:
Closing this due it being stale.

closed 2024-09-15 by AnonymousWP