Universal-Debloater-Alliance/universal-android-debloater-next-generation #583: feat(pkg-list): Add a `Safe` removal list

Rudxain, 2024-08-13:
### Describe the feature you want

https://discord.com/channels/1168607797081022514/1168607798196715573/1273024610035564595

We should split `Recommended` in 2 variants:
- Recommended: for outright malicious apps, or with known security/privacy vulnerabilities
- Safe: for potentially useful apps

## Alternative solution
We could move some packs from `Recommended` to `Advanced`, but that seems misleading, as removing `Recommended` packs should never break the system in any way

## Prior art
There's precedent for this: [uBO](https://github.com/gorhill/uBlock) does something similar: malware/badware & ads are filter-lists that are enabled by default, but "annoyances" is disabled by default because some users want what others consider "annoying"

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

jacklollz2, 2024-08-15:
I was just about to recommend this.
Some of these "Recommended" entries are not for the average user, like removing Gmail and such.

Rudxain, 2024-08-15:
I've been considering another alternative solution, which seems more complicated but could pay-off in the long run:

1. Move all `Advanced`s to `Expert`
2. Rename `Advanced` to `Safe`
3. Move all `Recommended`s to `Safe`
4. Gradually (opportunistically) move some `Safe` packs to `Recommend`

The 1st 3 steps can be automated, but the last one requires human work

Frigyes06, 2024-08-21:
I agree with this. I think these packages make it into recommended because most package list maintainers (rightfully) think that they are bloat/spyware. If I understand correctly, you're proposing a list that includes only packages the user is unlikely to interact with, therefore won't likely need, and only include packages that are highly unlikely to break any other functions?

Rudxain, 2024-08-21:
> If I understand correctly, you're proposing a list that includes only packages the user is unlikely to interact with, therefore won't likely need, and only include packages that are highly unlikely to break any other functions?

Yes! But those are just a few of many factors to take into account when deciding the recommendation category.

For example, a `Recommended` pack could be so malicious, that even if it does break the system when removed, it would be worth it. However, if the breakage is too much, then the "evilness" cancels out, so it should be in `Advanced` instead.

It could be said that my proposal/RFC only adds 1 or 2 more factors to decide the recommendation, but those factors were already being considered since a long time, so only the categories/recommendations would change

AnonymousWP, 2024-08-22:
This sounds like an okay feature, but we need to investigate how to implement this, because it's also prone to subjectiveness. Also, our top priority for now should be 1.1.0 and the current open PRs:

- https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/milestone/4
- https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pulls

Rudxain, 2024-09-22:
Should we close #640 as dupe of this?

crash5, 2024-09-23:
I would like to suggest a "name" key to store the app name in the uad file too. For example, the "Bolt: Request a Ride" app package name is "ee.mtakso.client". Which is pretty hard to figure out without additional googling. Some samsung packages has the same "sane" naming too :)

Rudxain, 2024-09-23:
> "name" key

I agree, but there's multiple names:
- The display name declared in `AndroidManifest.xml` of the version installed on each device (each device can have the same pack with a different name)
- The multiple alternative names in the `description` field, which may include inferred (unofficial) names (such as "vebgm" -> "Video Editor BackGround Music")

We still haven't found a proper way to show the display-name and app-icon of installed apps

AnonymousWP, 2024-10-04:
Will be done together with https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/discussions/646 and https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/discussions/608.

Rudxain, 2025-01-12:
I think we should transfer this issue to [the new repo](https://github.com/Universal-Debloater-Alliance/universal-android-preinstalled-lists/), as it's likely we'll migrate before doing this change

AnonymousWP, 2025-01-12:
> I think we should transfer this issue to [the new repo](https://github.com/Universal-Debloater-Alliance/universal-android-preinstalled-lists/), as it's likely we'll migrate before doing this change

But it's a pretty big change. It's part of the 2.0.0 milestone, so it'll take some time before we get there. But yeah, I definitely agree we should move this issue (along with other package-related issues) to that repo.

Rudxain, 2025-05-05:
> packages can have more than one tag

This is also the case for [IME](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/Glossary#IME)s:
- `Recommended` if you have another IME to fill its role
- `Advanced` if there are no other IMEs

This is why [ADL has a `replace` category](https://github.com/MuntashirAkon/android-debloat-list/blob/1ac05ca509b8a5849c3256b974f9fa1e00daf856/README.md#values-for-removal). I feel like `replace` would be very helpful

Muriilow, 2025-10-21:
Some notes: 
- The search is still working, even without the tag in the json. Is this updated? 
  - > labels: Will be removed in a future version, especially when [description search](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/184) is implemented. 

- I want to open a new branch that follows the [Trunk-Based Development](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/main/CONTRIBUTING.md) to pull this change, but before that i need my pull request to be checked so I dont have any problem with git versions and some opinions of what could change . #1113

Rudxain, 2025-10-21:
> changes the fields "Recommended" to "Safe", and "Expert" to "Disruptive".

I agree with this. It's a small step in the right direction!

> Expert and Advanced can have similar meanings so thats not good.

I also agree that there must be more distinction (both in principle and in practice), but if we re-add `Recommended` in the future, then it'd be nice if there were less `removal` levels to worry about. That's why I proposed merging `Advanced` with `Expert`, because having 2 "middle-levels" isn't useful enough (IMO, not speaking for the team)

> The search is still working, even without the tag in the json. Is this updated?

yeah, `labels` should've been removed long ago 😅

Muriilow, 2025-10-22:
> That's why I proposed merging Advanced with Expert, because having 2 "middle-levels" isn't useful enough (IMO, not speaking for the team) 

That's fair, I can do that. For now Im going to wait what everyone has to say and then we can continue!

kikecalpe, 2025-10-23:
My comment was long ago, I'll try to go back my mind to explain me.

> But instead of Untested or Expert I changed to Disruptive, to maintain the logic said in [FAQ](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ)

I don't understand this logic, but this is a language handicap.
I selected that words as a suggestion because are easy and clear concepts for me, but of course there should be most descriptive ones.
I came here because some tags were confusing and not enough descriptive. About "Disruptive", yes, I had to search for its meaning 😅, but if it sounds good for you and can synthesize that concept from the packages you want to group, you know what's better for UAD. 

> I think we already have a tool to say if something is malicious and that is the description, tags should have a unique meaning "Is it safe". Havin tags that is not close to this idea makes more confusing to the user and more complex to implement.

You cannot group together by description, as a point to have in mind.
I looked for easily group packages with common points, and mentioned "Malicious" as an example and because it makes sense for me and was mentioned before, not as a request. If you look exclusively for tags to group by different levels of safety, it's good the way you mentioned.