Universal-Debloater-Alliance/universal-android-debloater-next-generation #585: pkg(scope): Recommended debloat issue on OPPO

Aetumn12, 2024-08-16:
### Describe the problem

So i have an OPPO phone and i debloated all the apps in the recommended list only.The phone boots up fine until im brought upto the home screen and then the screen started flashing and phone becomes unresponsive

### You have a solution?

undo the debloat by performing factory reset

### Provide logs

[UAD_20240817.log](https://github.com/user-attachments/files/16639383/UAD_20240817.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-08-18:
Do you have an idea which package is responsible for this?

Asra05, 2024-09-30:
> Do you have an idea which package is responsible for this?

The package causing this problem on my device seems to be “com.coloros.weather.service”

AnonymousWP, 2024-10-04:
Uh-oh, that one is also present on my OnePlus Nord 3. I'll work on this.

closed 2024-10-27 by AnonymousWP