Universal-Debloater-Alliance/universal-android-debloater-next-generation #592: Missing bloatwares (Xiaomi Poco X6 Pro)

cretoo, 2024-08-24:
### Describe the bug

Some of the bloatwares are not listed in UAD, e.g. netflix, facebook, spotify music, the preinstalled game apps, etc.

### Expected behavior

Should be listed in UAD for uninstalling

### Operating System

Windows

### Provide logs

2024-08-25 00:18:49 DEBUG [src\core\update.rs:180] Checking for UAD-ng update
2024-08-25 00:18:49 INFO  [src\gui\views\list.rs:144] -----------------------------------------------------------------
2024-08-25 00:18:49 INFO  [src\gui\views\list.rs:145] ANDROID_SDK: 34 | DEVICE: POCO 2311DRK48G
2024-08-25 00:18:49 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------
2024-08-25 00:20:13 INFO  [src\gui\views\list.rs:144] -----------------------------------------------------------------
2024-08-25 00:20:13 INFO  [src\gui\views\list.rs:145] ANDROID_SDK: 34 | DEVICE: POCO 2311DRK48G
2024-08-25 00:20:13 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------
2024-08-25 00:21:24 ERROR [src\core\config.rs:79] Failed to read config file: `The system cannot find the file specified. (os error 2)`
2024-08-25 00:21:24 ERROR [src\core\config.rs:81] Restoring default config file
2024-08-25 00:21:24 DEBUG [src\core\update.rs:180] Checking for UAD-ng update
2024-08-25 00:21:24 INFO  [src\gui\views\list.rs:144] -----------------------------------------------------------------
2024-08-25 00:21:24 INFO  [src\gui\views\list.rs:145] ANDROID_SDK: 34 | DEVICE: POCO 2311DRK48G
2024-08-25 00:21:24 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------
2024-08-25 00:25:47 DEBUG [src\core\update.rs:180] Checking for UAD-ng update
2024-08-25 00:25:47 DEBUG [src\core\update.rs:204] Failed to check UAD-ng update
2024-08-25 00:26:11 DEBUG [src\core\update.rs:180] Checking for UAD-ng update
2024-08-25 00:26:23 INFO  [src\gui\views\list.rs:144] -----------------------------------------------------------------
2024-08-25 00:26:23 INFO  [src\gui\views\list.rs:145] ANDROID_SDK: 34 | DEVICE: POCO 2311DRK48G
2024-08-25 00:26:23 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------
2024-08-25 00:29:01 DEBUG [src\core\update.rs:180] Checking for UAD-ng update
2024-08-25 00:29:02 INFO  [src\gui\views\list.rs:144] -----------------------------------------------------------------
2024-08-25 00:29:02 INFO  [src\gui\views\list.rs:145] ANDROID_SDK: 34 | DEVICE: POCO 2311DRK48G
2024-08-25 00:29:02 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

cretoo, 2024-08-25:
Tried it on another phone (xiaomi poco m6 pro) and none of the bloatwares that i mentioned appeared too.

Rudxain, 2024-08-25:
That's weird. I remember an issue where no packs would show up in the GUI. This seems related

AnonymousWP, 2024-08-25:
We should pick this up if this really is a severe bug.

cretoo, 2024-08-27:
Is there any chance of this being fixed?

Rudxain, 2024-08-27:
@cretoo I guess most (all?) org members don't have multi-user devices, to test updates. So we'll have to redirect users (such as you) to the [unofficial "nightly" releases](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/actions) (actually, workflow runs)

Maybe the bug has been fixed on the `main` branch without anyone noticing? 👀

cretoo, 2024-08-30:
@Rudxain Thanks i'll take a look at it. It seems like UADNG only list system apps and not preinstalled user apps.

Rudxain, 2024-08-30:
> @Rudxain Thanks i'll take a look at it.

You're welcome! BTW, those releases are generally triggered manually, so even those might be "old" compared to `main`.

> preinstalled user apps.

Such as system apps that got updated? 👀

cretoo, 2024-08-30:
> Such as system apps that got updated?

I don't think so, it's a brand new phone out of the box. I was able to see the mentioned apps (Netflix, facebook, etc.) listed under user apps on ADB AppControl which is another debloater, so i thought UADNG only list system apps and not user apps.

Rudxain, 2024-08-30:
> only list system apps and not user apps.

Similar (old) report: 0x192/universal-android-debloater#795

It seems like an ADB bug. Or maybe the OEM has hidden those packs from ADB for some reason? It seems plausible, considering Samsung Galaxy "Deep Sleeping" apps are considered "Disabled" by the Android system

AnonymousWP, 2024-09-15:
So I guess we can close this issue then considering both things are out of our control.

Rudxain, 2024-09-16:
@cretoo run this cmd:
```sh
adb shell pm list packages -3
```
> [!note]
> for privacy, don't post the output

Does it print the missing packs?

If not, try this:
```sh
adb shell pm list packages `-a`
```
> [!warning]
> DO NOT POST THE FULL OUTPUT!

If any of those work, then we can fix it on our side, but it has to be a new toggle setting. The `-a` flag is very verbose, so it'll be annoying for some users.

I suspect the OEM has misleadingly marked as "user" some "system" packs, that's why `pm list packages -s` is so empty

Rudxain, 2024-09-16:
huh, I thought `-a` was the default. This proves me wrong:
```sh
pm list packages -a | wc -l # 399
pm list packages | wc -l # 237
```

Frigyes06, 2024-10-07:
Uad only lists system packages. Some phones have some apps as system vs third party. These apps are usually easily deletable from the device UI, however.

Rudxain, 2024-12-16:
Related #607

zalnaRs, 2025-01-02:
I have a Redmi Note 13 Pro 5G and the POCO X6 5G both are the same phones. I see all of them

AnonymousWP, 2025-01-06:
Not sure what we should do with this issue now to be honest. As Frigyes said: you can easily uninstall/remove pre-installed user-apps easily.

Rudxain, 2025-01-07:
IMO, we should simply show all packs (as an opt-in), not just `-s`. But this is far from a priority, unless there's a device model where at least 1 pack in `-3` cannot be disabled/uninstalled

AnonymousWP, 2025-01-20:
Closing this issue because of Frigyes' and my conclusion. We can always re-open it later, but it's by design (you can just uninstall user-installed apps by yourself).

closed 2025-01-20 by AnonymousWP

Rudxain, 2025-01-20:
I agree, but

> uninstall user-installed apps

not all `-3` are manually-installed by the user 😕. Yes, they can be disabled by ADB and Settings-app, but what about the other features UAD-NG provides? (description, removal-type, backups, etc...)