Universal-Debloater-Alliance/universal-android-debloater-next-generation #594: pkg(scope): Number of apps missing for Motorola Edge (2024)

Lost-Entrepreneur439, 2024-08-28:
### List your packages

APP 1:
- Package name: com.android.internal.systemui.navbar.hidegestural
- List: AOSP
- Removal: Expert
- Description: This package handles the internal management of gesture-based navigation bars in the Android system UI. Removing it may disrupt the functionality of gesture navigation, which is integral to the user interface.

APP 2:
- Package name: com.android.internal.systemui.navbar.transparent
- List: AOSP
- Removal: Advanced
- Description:This package is responsible for managing the transparency of the navigation bar in the Android system UI. It controls how the navigation bar appears in terms of opacity, allowing it to blend with the background or be more visible. Removing this package may affect the visual appearance of the navigation bar, potentially leading to a less customizable or aesthetically inconsistent UI.

APP 3:
- Package name: com.android.phone.overlay.motcommon
- List: OEM
- Removal: Advanced
- Description: This package is part of Motorola's custom overlay for phone-related features and UI elements. It is likely involved in providing specific functionality or visual customization for phone operations on Motorola devices. Removing this package could affect certain phone features or the look and feel of phone-related settings specific to Motorola devices.

APP 4:
- Package name: com.android.systemui.accessibility.accessibilitymenu.auto_generated_rro_product__
- List: AOSP
- Removal: Advanced
- Description: This package is associated with the Accessibility Menu in Android's system UI, specifically related to auto-generated resources for the menu. It helps manage accessibility features that assist users with disabilities. Removing this package might disable or alter accessibility options provided through the Accessibility Menu, potentially affecting users who rely on these features.

APP 5:
- Package name: com.android.wifi.resources.overlay.motCommon
- List: OEM
- Removal: Advanced
- Description: This package is part of Motorola's custom overlay for Wi-Fi settings and resources. It includes Motorola-specific modifications or enhancements for Wi-Fi management and user interface elements related to Wi-Fi on Motorola devices. Removing this package might affect certain Motorola-specific Wi-Fi features or customizations, potentially leading to changes in how Wi-Fi settings are displayed or managed.

APP 6:
- Package name: com.android.wifi.resources.overlay.WifiResScanCountryCode
- List: AOSP
- Removal: Advanced
- Description: This package is related to Wi-Fi resources and overlays, specifically dealing with country code scanning for Wi-Fi networks. It manages resources that could influence how Wi-Fi scans are handled based on the country code settings. Removing this package might impact the handling of Wi-Fi scanning processes, particularly those affected by regional settings, but it is unlikely to disrupt core Wi-Fi functionality.

APP 7:
- Package name: com.google.android.health.connect.backuprestore
- List: Google
- Removal: Recommended
- Description: This package is related to Google Health Connect and handles the backup and restore functionality for health-related data. Removing this package could impact the ability to back up and restore health data associated with Google Health Connect.

APP 8:
- Package name: com.google.android.healthconnect.controller
- List: Google
- Removal: Recommended
- Description: This package is part of Google Health Connect and is responsible for managing and controlling health-related data interactions. It handles data permissions, syncing, and integration between different health-related apps and services.

APP 9:
- Package name: com.google.android.wifi.dialog
- List: Google
- Removal: Recommended
- Description: This package is associated with the user interface elements related to Wi-Fi management in Android. It handles the dialogs and prompts that appear when managing Wi-Fi connections, such as network selection, Wi-Fi settings, and connectivity prompts. Removing this package might impact the appearance or functionality of Wi-Fi management dialogs, potentially making it harder to interact with Wi-Fi settings or connect to networks.

APP 10:
- Package name: com.google.android.wifi.resources
- List: Google
- Removal: Advanced
- Description: This package contains resources related to Wi-Fi management, including string resources, images, and layout files used in the Wi-Fi settings and user interface. It helps provide the necessary resources for displaying and interacting with Wi-Fi settings in the system UI. Removing this package might affect the appearance or usability of Wi-Fi settings and related features, potentially leading to missing resources or visual inconsistencies in Wi-Fi management interfaces.

APP 11:
- Package name: com.motorola.android.coresettingsext.overlay.avatrn
- List: OEM
- Removal: Expert
- Description: This package is part of Motorola's custom overlay for system settings, specifically tailored for the Motorola Edge 2024 (codename "avatrn"). It includes device-specific settings or customizations that are unique to this model. Removing this package could significantly impact Motorola-specific features and custom settings on the Edge 2024, potentially leading to issues with functionality or user interface elements that are specific to this device model.

APP 12:
- Package name: com.motorola.android.launcher.overlay.motolauncheroverlayna
[... 123 lines cut ...]

APP 33:
- Package name: com.motorola.setup.auto_generated_rro_product__
- List: OEM
- Removal: Recommended
- Description: This package is involved in Motorola's setup process, particularly handling automatically generated resources related to the setup of the device. It includes Motorola-specific resources that are used during the initial setup phase. Since this package is primarily needed for the setup process and will be restored after a factory reset, it is not critical for ongoing functionality.

APP 34:
- Package name: com.motorola.setup.overlay.dataenable
- List: OEM
- Removal: Recommended
- Description: This package is used during the Motorola device setup process, specifically for managing or presenting settings related to mobile data. It includes Motorola-specific customizations or UI elements that assist in configuring mobile data during the initial setup. Since it is only needed for the setup process and will be restored upon a factory reset, removing this package will not impact the core functionality of the device.

APP 35:
- Package name: com.oem.euiccpartnerapp.overlay.retca
- List: OEM
- Removal: Advanced
- Description: This package is related to the eUICC (embedded Universal Integrated Circuit Card) partner application, specifically an OEM overlay that deals with eSIM functionality and management. The "retca" suffix suggests it could be tied to a specific regional or carrier customization. It likely includes additional features or user interface elements specific to the management and configuration of eSIMs, provided by Motorola or in collaboration with a carrier. Removing this package might impact eSIM management features, potentially affecting the user experience related to eSIM provisioning and configuration.

### Acknowledgements

- [X] I removed all those packages on my phone. If not, why? Explain in the issue.
- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-08-30:
@Lost-Entrepreneur439 If you can, could you please add them by yourself as a PR? That'd save us a lot of work and is easier to review. Thanks for the extensive issue!

Lost-Entrepreneur439, 2024-09-01:
> @Lost-Entrepreneur439 If you can, could you please add them by yourself as a PR? That'd save us a lot of work and is easier to review. Thanks for the extensive issue!

I will work on it, however my computers motherboard has failed so I am quite limited in stuff I am able to do as of right now.

AnonymousWP, 2024-09-01:
> > @Lost-Entrepreneur439 If you can, could you please add them by yourself as a PR? That'd save us a lot of work and is easier to review. Thanks for the extensive issue!
> 
> I will work on it, however my computers motherboard has failed so I am quite limited in stuff I am able to do as of right now.

That'd be great, but good luck with your motherboard. 🙂

PR #610 (2024-09-14): pkg(motorola): add packages
This closes #594
closed 2024-09-14 by AnonymousWP