Universal-Debloater-Alliance/universal-android-debloater-next-generation #595: pkg(scope): Notifications not showing on turned off screen

smoreno12, 2024-08-28:
### Describe the problem

After debloating Samsung A35, notifications won't appear when screen is fully turned off (aod thing i guess, notification just pops up on upper part of screen, rest of screen remains dark).

### Provide logs

.

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-08-28:
Can you tell us which app is responsible for this?

smoreno12, 2024-08-28:
Couldn't find it, I uninstalled 200 apps at once

AnonymousWP, 2024-08-28:
You shouldn't disable/uninstall so many packages at once lol. Read the description, or better: check out which list you're using.

Rudxain, 2024-08-28:
I suspect the package must be some AOSP or Google stuff. Samsung probably isn't related (unless it's the "Push" app)

schbrns, 2024-11-24:
Feel free to re-open this if you have any issues regarding packages that you can identify. Thanks.

closed 2024-11-24 by schbrns