Universal-Debloater-Alliance/universal-android-debloater-next-generation #598: bug(scope): no app_id on wayland

deathtrip, 2024-08-31:
### Describe the bug

When running UAD-ng, or the previous gen UAD for that matter, the app_id value on wayland is empty.
This is a problem when setting window rules on tiling window managers, and for programs thet rename workspaces based on their app_id.

### Expected behavior

There should be a app_id value.

### Operating System

Linux

### Window Manager

sway

### Provide logs

I don't think it applies to this issue.

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-09-01:
I guess this is more like a feature request? [Iced supports this by default](https://github.com/iced-rs/iced/pull/599). This might get resolved by #551

deathtrip, 2024-09-01:
It's not really a feature, more like a standard, and it's a rarity for an application to not provide it.

AnonymousWP, 2024-09-01:
Could you help us by submitting a PR?

deathtrip, 2024-09-03:
Unfortunately not, as i am not familiar with the UAD codebase nor rust in general.

Rudxain, 2024-09-04:
I was reading [the docs](https://docs.rs/iced/0.12.1/iced/settings/struct.Settings.html), and it turns out (according to `rust-analyzer`) we aren't even touching `iced::settings` 💀

I'll open a PR for that, but I still have no idea where to pass the `struct`

Rudxain, 2024-09-04:
Wait, **we are using** `iced::Settings` but not `iced::settings::Settings`

Rudxain, 2024-09-04:
Wait, **we are using** `iced::Settings` (alias of `iced::settings::Settings`), but not `iced::settings` directly

Rudxain, 2024-09-04:
Found it!:
https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/4f9332c2c29d688cae832e8de8bf400f07b5eb4c/src/gui/mod.rs#L390-L403

It's an easy fix! We just have to specify the `id` field, because `default` is `None` (the Iced PR made me believe it was inferred from crate name, which is what [`Clap`](https://docs.rs/clap/latest/clap/index.html) does)

Rudxain, 2024-09-04:
@deathtrip Can you please test [this version](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/actions/runs/10696660871) to see if the metadata is set?

I have Wayland, but not Sway, only Mutter

deathtrip, 2024-09-08:
Just tested the version linked in the above post, and the app_id is still empty.

Rudxain, 2024-09-09:
After reading more docs, I suspect this is an Iced bug, which implies this isn't something we can/should fix. Unless, it's something specific to the version `main` uses, in that case it could be fixed by #551

PR #603 (2024-09-04): refactor(app_id): specify `iced::Settings.id`
Closes #598

I decided to do an opportunistic refactoring of the app name, because this cmd:
```sh
grep -ri '"[^"]*UAD-ng[^"]*"' src/
```
printed too many lines