Universal-Debloater-Alliance/universal-android-debloater-next-generation #599: bug(scope): unable to run program

kattsukii, 2024-09-02:
### Describe the bug

Download the latest version, then run it.

the message i get after i run it

```
2024-09-01 23:38:40 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:38:40 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
wl_surface#22: error 0: DRM Syncobj surface object already created for surface 22
Protocol error 0 on object wl_surface@22: 
Protocol error 0 on object wl_surface@22: 
2024-09-01 23:38:40 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-01 23:38:40 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
thread 'main' panicked at /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-core-0.19.3/src/device/global.rs:1859:21:
internal error: entered unreachable code: Fallback system failed to choose present mode. This is a bug. Mode: AutoVsync, Options: []
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

### Expected behavior

to run

### Operating System

Linux

### Window Manager

Wayland

### Desktop Environment

GNOME 46.4, fedora 40

### Provide logs

2024-09-01 23:25:34 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:25:34 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:25:34 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-01 23:25:34 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-01 23:25:34 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
2024-09-01 23:25:46 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:25:46 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:25:47 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-01 23:25:47 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-01 23:25:47 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
2024-09-01 23:25:57 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:25:57 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:25:57 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-01 23:25:57 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-01 23:25:57 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
2024-09-01 23:27:09 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:27:09 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:27:09 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-01 23:27:09 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-01 23:27:09 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
2024-09-01 23:28:41 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:28:41 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:28:41 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-01 23:28:41 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-01 23:28:41 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
2024-09-01 23:30:29 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:30:29 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:30:29 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-01 23:30:29 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-01 23:30:29 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
2024-09-01 23:30:56 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:30:56 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:30:56 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
[... 3 lines cut ...]
2024-09-01 23:33:46 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-09-01 23:33:47 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-01 23:33:47 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
[... cut ...]
2024-09-01 23:38:40 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-01 23:38:40 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

theskpaul, 2024-09-02:
I recently compiled it on my system, and it works perfectly, as you can see in this picture. I'm running it on Plasma 6 Wayland [Arch Linux].
I think Vulkan drivers are missing from your system. Check it first! [image]

AnonymousWP, 2024-09-02:
Thanks for the explanation and try-out, [theskpaul](https://github.com/theskpaul). 🙂

cone-del, 2024-09-04:
I had same problem in Windows 11, with an 
"INVALID PARAMETER", only .000040 sec execution time 

It does need remote SSL inquires validation to checksum? because i got problem with openssl SSL validation

rscm, 2024-09-04:
I have the exact same issue

OS: Fedora 40 Silverblue

```
❯ rpm-ostree status -b
State: idle
BootedDeployment:
● fedora:fedora/40/x86_64/silverblue
                  Version: 40.20240902.0 (2024-09-02T00:42:54Z)
               BaseCommit: 14a2faa512f11119ddbf672b4014c55a2658498be14d44812a47fcb663c4b7e8
             GPGSignature: Valid signature by 115DF9AEF857853EE8445D0A0727707EA15B79CC
      RemovedBasePackages: firefox firefox-langpacks 129.0.2-1.fc40
          LayeredPackages: akmod-nvidia android-tools distrobox gnome-tweaks gparted kernel-tools kitty printer-driver-brlaser rpmfusion-free-release rpmfusion-nonfree-release starship steam-devices tailscale
                           xboxdrv xorg-x11-drv-nvidia xorg-x11-drv-nvidia-cuda
```

this is from `$HOME/.cache/uad/UAD_20240904.log`

```
2024-09-04 18:38:13 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-04 18:38:13 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-04 18:38:13 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
2024-09-04 18:40:56 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-04 18:40:56 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-04 18:40:56 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
2024-09-04 18:43:12 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:268] No config found!
2024-09-04 18:43:12 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2024-09-04 18:43:12 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-04 18:43:12 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-04 18:43:12 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
2024-09-04 18:44:20 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:268] No config found!
2024-09-04 18:44:20 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2024-09-04 18:44:20 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-04 18:44:20 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-04 18:44:20 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
2024-09-04 18:44:37 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-04 18:44:37 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-04 18:44:37 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
2024-09-04 18:44:56 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-09-04 18:44:56 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2024-09-04 18:44:56 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR

```

I ran it using the iGPU and then dGPU (just in case)

Rudxain, 2024-09-05:
@rscm I collapsed `vulkaninfo` because it was **extremely long!**

Thank you for the debug info, but it was hard to scroll this page 😅

AnonymousWP, 2024-09-05:
Could this be related to the back-end that `iced` uses? Could you try compiling it with a new version of `iced`?

rscm, 2024-09-05:
> @rscm I collapsed `vulkaninfo` because it was **extremely long!**
> 
> Thank you for the debug info, but it was hard to scroll this page 😅

sorry I didn't knew that markdown syntax. I will use it from now on. Sorry again.

Rudxain, 2024-09-06:
It seems it's a [GH MD "flavor" non-standard HTML extension](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/organizing-information-with-collapsed-sections) (such as `<img>`)

Rudxain, 2024-09-14:
> related to the back-end that `iced` uses?

Maybe: iced-rs/iced#2572

Rudxain, 2024-09-14:
> related to the back-end that `iced` uses?

Maybe:
- iced-rs/iced#2562

AnonymousWP, 2024-09-15:
Hmm, I see that users say that it works with `wgpu` removed: https://github.com/iced-rs/iced/issues/2562#issuecomment-2322787586. Should we offer builds with and without `wgpu` if that's possible?

Rudxain, 2024-09-16:
related #556

Rudxain, 2024-09-16:
> builds with and without `wgpu` if that's possible?

We should try:
1. create a branch that disables `wgpu`
2. generate build-artifacts from the branch
3. ask users to try running those artifacts

Rudxain, 2024-09-18:
https://github.com/iced-rs/iced/blob/f89744283167a1961fcff512ad48b0eb9b8fcaef/.github/ISSUE_TEMPLATE/BUG-REPORT.yml?plain=1#L39-L48

Nj0be, 2024-10-10:
I have the same issue on arch linux. This is the log:
```
2024-10-10 21:12:45 WARN  [/home/benia/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
2024-10-10 21:12:45 WARN  [/home/benia/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/adapter.rs:774] Detected skylake derivative running on mesa i915. Clears to srgb textures will use manual shader clears.
wp_linux_drm_syncobj_manager_v1#61: error 0: Surface already has a syncobj attached
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61: 
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61: 
Protocol error 2024-10-10 21:12:46 ERROR [/home/benia/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
0 on object wp_linux_drm_syncobj_manager_v1@61: 
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61: 
2024-10-10 21:12:46 ERROR [/home/benia/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@thread '61main: ' panicked at 
/home/benia/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-core-0.19.4/src/device/global.rsProtocol error :01859 on object :wp_linux_drm_syncobj_manager_v121@:
61internal error: entered unreachable code: Fallback system failed to choose present mode. This is a bug. Mode: AutoVsync, Options: []: 

note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61: 
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61: 
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61: 
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61: 
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61: 
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61: 
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61: 
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61: 
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61: 
Protocol error 0 on object wp_linux_drm_syncobj_manager_v1@61:
```

rushmash, 2024-11-05:
I had the same issue on old NVIDIA drivers. After updating to 565.57.01 the issue is gone.
Can anybody confirm?

Lost-Saint, 2024-11-06:
yep it works on the new nvidia drivers 
version: 565.57.01

rushmash, 2024-11-10:
@brucetheelephant You won't fix it without updating anything...

@Lost-Saint Since the app works on new drivers, can you close the issue?

closed 2024-11-10 by Lost-Saint