Universal-Debloater-Alliance/universal-android-debloater-next-generation #601: bug(scope): ADB is installed and running, detects S24 Ultra but debloater app says ADB is not installed

gsadasivan, 2024-09-02:
### Describe the bug

ADB is installed and running, detects S24 Ultra but debloater app says ADB is not installed
Windows 11
[image]

### Expected behavior

My device shuld be found.

### You have a solution?

Even though ADB directory was in my path, the program failed to detect my device. However, if I moved the program to the ADB directory, and launched it, it seems to work.

### Operating System

Windows

### Provide logs

[UAD_20240902.log](https://github.com/user-attachments/files/16841396/UAD_20240902.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-09-03:
Does your PC have 2 storage media installed? I suspect that whatever library UAD-NG is using, may not take into account secondary storage devices when searching `PATH`

sudo-userv2, 2024-09-20:
Try running this as administrator : ` adb start-server`

ItsEagle, 2024-10-14:
you have to disable and reenable usb debugging on your s24 in the developer settings while being connected to your pc

AnonymousWP, 2024-10-18:
Solution has been given and user did not reply, so I'm closing this issue.

closed 2024-10-18 by AnonymousWP