Universal-Debloater-Alliance/universal-android-debloater-next-generation #602: pkg(scope): Poco X6 Pro functions missing after debloating (Photo Cut out and Wallpaper settings)

pacopoco, 2024-09-04:
### Describe the problem

Hi,
so i used the debloater in the recommended, enabled, all lists mode and uninstalled 133 apps. i uninstalled some before on the phone itself with the deepcleaner. 
Now i cant cut out (by touching and holding) and copy photos in my gallery anymore which definetly worked before (new feature on HyperOS on my Poco X6 Pro (similar to iOS).
Also i cant touch and hold on the lockscreen to open and edit my wallpaper which i also miss.. was cool features. 
im a little lost, i tried reinstalling some apps which i thought would work but none did. If you could help me i would be very glad :) 

Thank you :) 

### Provide logs

2024-09-04 01:25:35 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-core-0.19.3\src\instance.rs:342] Missing downlevel flags: DownlevelFlags(SURFACE_VIEW_FORMATS)
The underlying API or device in use does not support enough features to be a fully compliant implementation of WebGPU. A subset of the features can still be used. If you are running this program on native and not in a browser and wish to limit the features you use to the supported subset, call Adapter::downlevel_properties or Device::downlevel_properties to get a listing of the features the current platform supports.
2024-09-04 01:25:35 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-core-0.19.3\src\instance.rs:347] DownlevelCapabilities {
    flags: DownlevelFlags(
        COMPUTE_SHADERS | FRAGMENT_WRITABLE_STORAGE | INDIRECT_EXECUTION | BASE_VERTEX | READ_ONLY_DEPTH_STENCIL | NON_POWER_OF_TWO_MIPMAPPED_TEXTURES | CUBE_ARRAY_TEXTURES | COMPARISON_SAMPLERS | INDEPENDENT_BLEND | VERTEX_STORAGE | ANISOTROPIC_FILTERING | FRAGMENT_STORAGE | MULTISAMPLED_SHADING | DEPTH_TEXTURE_AND_BUFFER_COPIES | WEBGPU_TEXTURE_FORMAT_SUPPORT | BUFFER_BINDINGS_NOT_16_BYTE_ALIGNED | UNRESTRICTED_INDEX_BUFFER | FULL_DRAW_INDEX_UINT32 | DEPTH_BIAS_CLAMP | VIEW_FORMATS | UNRESTRICTED_EXTERNAL_TEXTURE_COPIES | NONBLOCKING_QUERY_RESOLVE | VERTEX_AND_INSTANCE_INDEX_RESPECTS_RESPECTIVE_FIRST_VALUE_IN_INDIRECT_DRAW,
    ),
    limits: DownlevelLimits,
    shader_model: Sm5,
}
2024-09-04 01:25:35 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\fontdb-0.15.0\src\lib.rs:241] Failed to load a font face 0 from 'C:\Windows\Fonts\mstmc.ttf' cause malformed font.
2024-09-04 01:25:35 DEBUG [src\core\update.rs:180] Checking for UAD-ng update
2024-09-04 01:25:40 INFO  [src\gui\views\list.rs:144] -----------------------------------------------------------------
2024-09-04 01:25:40 INFO  [src\gui\views\list.rs:145] ANDROID_SDK: 34 | DEVICE: POCO 2311DRK48G
2024-09-04 01:25:40 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------
2024-09-04 01:26:32 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.cameraextensions -> Package com.android.cameraextensions installed for user: 0
2024-09-04 01:27:30 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.systemui.overlay.miui -> Package com.android.systemui.overlay.miui installed for user: 0
2024-09-04 01:29:12 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.android.systemui.overlay.miui -> Success
2024-09-04 01:41:13 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.miui.miwallpaper.overlay -> Package com.miui.miwallpaper.overlay installed for user: 0
2024-09-04 01:41:14 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.miui.msa.global -> Package com.miui.msa.global installed for user: 0
2024-09-04 01:41:21 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.miui.miwallpaper.overlay.customize -> Package com.miui.miwallpaper.overlay.customize installed for user: 0
2024-09-04 01:42:01 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.miui.qr -> Package com.miui.qr installed for user: 0
2024-09-04 01:42:39 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.miui.wallpaper.overlay -> Package com.miui.wallpaper.overlay installed for user: 0
2024-09-04 01:42:45 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.miui.wallpaper.overlay.customize -> Package com.miui.wallpaper.overlay.customize installed for user: 0
2024-09-04 01:43:41 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.xiaomi.cameratools -> Package com.xiaomi.cameratools installed for user: 0
2024-09-04 01:45:27 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.google.android.apps.maps -> Success
2024-09-04 01:46:01 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.miui.msa.global -> Success
2024-09-04 01:46:23 ERROR [src\core\sync.rs:129] [Shell] reboot -> 
2024-09-04 02:02:56 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-core-0.19.3\src\instance.rs:342] Missing downlevel flags: DownlevelFlags(SURFACE_VIEW_FORMATS)
The underlying API or device in use does not support enough features to be a fully compliant implementation of WebGPU. A subset of the features can still be used. If you are running this program on native and not in a browser and wish to limit the features you use to the supported subset, call Adapter::downlevel_properties or Device::downlevel_properties to get a listing of the features the current platform supports.
2024-09-04 02:02:56 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-core-0.19.3\src\instance.rs:347] DownlevelCapabilities {
    flags: DownlevelFlags(
        COMPUTE_SHADERS | FRAGMENT_WRITABLE_STORAGE | INDIRECT_EXECUTION | BASE_VERTEX | READ_ONLY_DEPTH_STENCIL | NON_POWER_OF_TWO_MIPMAPPED_TEXTURES | CUBE_ARRAY_TEXTURES | COMPARISON_SAMPLERS | INDEPENDENT_BLEND | VERTEX_STORAGE | ANISOTROPIC_FILTERING | FRAGMENT_STORAGE | MULTISAMPLED_SHADING | DEPTH_TEXTURE_AND_BUFFER_COPIES | WEBGPU_TEXTURE_FORMAT_SUPPORT | BUFFER_BINDINGS_NOT_16_BYTE_ALIGNED | UNRESTRICTED_INDEX_BUFFER | FULL_DRAW_INDEX_UINT32 | DEPTH_BIAS_CLAMP | VIEW_FORMATS | UNRESTRICTED_EXTERNAL_TEXTURE_COPIES | NONBLOCKING_QUERY_RESOLVE | VERTEX_AND_INSTANCE_INDEX_RESPECTS_RESPECTIVE_FIRST_VALUE_IN_INDIRECT_DRAW,
    ),
    limits: DownlevelLimits,
    shader_model: Sm5,
}
2024-09-04 02:02:57 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\fontdb-0.15.0\src\lib.rs:241] Failed to load a font face 0 from 'C:\Windows\Fonts\mstmc.ttf' cause malformed font.
2024-09-04 02:02:57 DEBUG [src\core\update.rs:180] Checking for UAD-ng update
2024-09-04 02:03:03 INFO  [src\gui\views\list.rs:144] -----------------------------------------------------------------
2024-09-04 02:03:03 INFO  [src\gui\views\list.rs:145] ANDROID_SDK: 34 | DEVICE: POCO 2311DRK48G
2024-09-04 02:03:03 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------
2024-09-04 02:03:21 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.android.cameraextensions -> Success
2024-09-04 02:03:32 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.miui.miwallpaper.overlay -> Success
2024-09-04 02:03:35 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.miui.miwallpaper.overlay.customize -> Success
2024-09-04 02:03:44 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.miui.wallpaper.overlay.customize -> Success
2024-09-04 02:03:47 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.miui.wallpaper.overlay -> Success
2024-09-04 02:09:00 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.as -> Package com.google.android.as installed for user: 0
2024-09-04 02:10:41 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.mediatek.miravision.ui -> Package com.mediatek.miravision.ui installed for user: 0
2024-09-04 02:12:16 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.xiaomi.cameratools -> Success
2024-09-04 02:12:27 ERROR [src\core\sync.rs:129] [Shell] reboot -> 
2024-09-04 02:17:25 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-core-0.19.3\src\instance.rs:342] Missing downlevel flags: DownlevelFlags(SURFACE_VIEW_FORMATS)
[... cut ...]
2024-09-04 02:17:31 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------
2024-09-04 02:17:36 DEBUG [src\core\update.rs:180] Checking for UAD-ng update
2024-09-04 02:17:49 ERROR [src\core\utils.rs:78] Can't open the following URL: 
2024-09-04 02:31:07 DEBUG [src\core\update.rs:180] Checking for UAD-ng update
2024-09-04 02:31:12 ERROR [src\core\utils.rs:78] Can't open the following URL: 

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-09-04:
I don't know, but it's a bad idea to mass-uninstall so many packages at the same time without reading the descriptions. You can try restoring all packages.

pacopoco, 2024-09-05:
Thank you :)Everything else seems to be working fine. Deepcleaner Option is also gone. I read all the descriptions while trying to Reinstall some but they all recommended uninstalling them. Maybe it came from the Apps i uninstalled before on the phone itself using deepcleaner.
Im worried for Security of data reinstalling them all but well i need to try i guess. If anybody knows the apps i need please reach Out.

AnonymousWP, 2024-09-06:
Yes, let us know. If you suspect it's an application that we've placed under `Recommended`, please open an issue for improving its documentation. Closing this issue for now.

closed 2024-09-06 by AnonymousWP