Universal-Debloater-Alliance/universal-android-debloater-next-generation #604: pkg(scope): Oneplus 9 Pro, Expert debloat, can send SMS but not receive

SpookySanta, 2024-09-04:
### Describe the problem

Have tried with two different SMS apps.

### Provide logs

[log9pro.txt](https://github.com/user-attachments/files/16880744/log9pro.txt)

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-09-04:
Why did you (seemingly blindly) uninstall so many advanced packages? Have you read the descriptions?

SpookySanta, 2024-09-04:
Yes, tought the expert option didn't break anything.

AnonymousWP, 2024-09-04:
Well, try to re-install everything. Next time, read it properly and read through https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ. Doesn't `expert` sound like it's far from `recommended` though? 😅

schbrns, 2024-11-24:
Feel free to re-open the issue if you have any questions regarding specific packages that you can identify. Thanks.

closed 2024-11-24 by schbrns