Universal-Debloater-Alliance/universal-android-debloater-next-generation #605: feat(accessibility): key-nav & touch-scrolling

Rudxain, 2024-09-04:
### Describe the feature you want

Support for keyboard navigation is non-existent: the arrow keys can't be used to scroll the pack list and `Tab` can't focus on any widget.

Surprisingly, touchscreen scrolling works (at least on Wayland+Mutter), but only when dragging the _background_ of the pack list (a tiny margin between list items; hard to "aim"). If UAD-ng somehow falls-back to x11, then this bug may be a limitation of `xwayland`.

Touchscreen clicks work perfectly fine

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-09-04:
I tested the touchscreen on a Linux system that mainly uses `libinput`, more specifically Debian 13 Trixie (not yet stable)

Rudxain, 2025-01-01:
@AnonymousWP should we split this in 2 feature requests? I guess it makes sense, because touch-scrolling isn't the only kind of touch-navigation (we could add "gestures" in the far future)

Rudxain, 2025-01-06:
I've noticed something weird (but good): 2-finger scrolling (touchpad) always works, it doesn't depend on pointer/cursor position. Nice workaround [for this](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/642#issuecomment-2366915447)

Rudxain, 2025-01-12:
I mean opening another for touch-screens, then closing this one

AnonymousWP, 2025-01-23:
That depends on whether the implementations for both touchscreens and key-nav are very different or not. If they require a very different approach, then it should be split. If it can be accomplished by using the same package/library, then not I'd say. But then again, I don't have experience regarding implementing key-nav & touch-scrolling features.