Universal-Debloater-Alliance/universal-android-debloater-next-generation #607: bug(scope): uad-ng-generated list of my apps is missing some of them

asakura42, 2024-09-05:
### Describe the bug

Thanks to uad-ng I disabled many apps, but apps like `com.miui.notes` and `com.miui.compass` do not show up in the list and are not removed automatically despite being present in the json file (I checked this json file in `~/.cache/uad/uad_lists.json`).

I checked some outputs for grepd `com.miui.notes` and compared with for example `com.xiaomi.xmsf`.

```bash
$ adb shell pm list packages -f | grep -i com.miui.notes
package:/data/app/~~SOME_STRING==/com.miui.notes-SOME_STRING==/base.apk=com.miui.notes
```

```bash
$ adb shell pm list packages -f | grep -i com.xiaomi.xmsf
package:/data/app/~~SOME_STRING==/com.xiaomi.xmsf-SOME_STRING==/base.apk=com.xiaomi.xmsf
```

These lines look similar.

Json structure is similar too :

```json
  "com.xiaomi.xmsf": {
    "list": "Oem",
    "description": "Xiaomi Service Framework\nContains a set of API's for Xiaomi apps. Expect widespread breakage of Xiaomi apps/functionality if disabled.\nDisabling will mess with Alarm clock functionality(according to issue#136) and break Mi Cloud and Mi account (and all features that depend on them).\nI don't know about now, but in 2016 this app constantly tried to establish tcp connections in the background.",
    "dependencies": [],
    "neededBy": [],
    "labels": [],
    "removal": "Expert"
  },
```

and

```json
  "com.miui.notes": {
    "list": "Oem",
    "description": "Mi Notes\n",
    "dependencies": [],
    "neededBy": [],
    "labels": [],
    "removal": "Recommended"
  },
```

Also I checked how many packages weren't disabled (and selected in uad-ng by default), there are 17:

```
com.xiaomi.scanner
com.miui.weather2
com.mi.global.pocobbs
ru.yandex.yandexmaps
com.miui.android.fashiongallery
ru.yandex.searchplugin
com.miui.screenrecorder
com.android.soundrecorder
com.xiaomi.midrop
com.duokan.phone.remotecontroller
com.miui.compass
com.miui.calculator
com.miui.notes
cn.wps.xiaomi.abroad.lite
com.yandex.browser
com.miui.huanji
com.mi.global.pocostore
```

This list was created with small script:

```bash
while IFS= read -r line ; do
jq -r 'to_entries | map(select(.value.removal == "Recommended") | .key) | .[]' ~/.cache/uad/uad_lists.json | grep "^$line$"
done <<< $(adb shell pm list packages -e | awk -F':' '{print $2}')
```

### Expected behavior

Possibility to select and delete/disable some OEM apps like `com.miui.notes`.

### You have a solution?

Manual uninstallation of these apps after uad-ng /shrug

### Operating System

Linux

### Window Manager

xorg

### Desktop Environment

No DE. WM is dwm.

### Provide logs

https://c-v.sh/sprightfulargent.txt

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-09-11:
Did you already solve this issue by yourself perhaps? It could be that on some devices it's a system app and on other's it's not.

asakura42, 2024-09-11:
@AnonymousWP I just disabled these apps manually.

Anyway, "system" Google packages (Services, Chrome etc) uad-ng detects and disables like a charm. But these packages - not.

AnonymousWP, 2024-09-13:
Do you believe this is a bug or due to what I suggested above? In case of we latter: we can close it.

Frigyes06, 2024-11-26:
> @AnonymousWP I just disabled these apps manually.
> 
> Anyway, "system" Google packages (Services, Chrome etc) uad-ng detects and disables like a charm. But these packages - not.

UAD-ng only works with system packages, specifically ones that present as system packages through ADB. We have had many conversations about this, and the decision always has been to have it only work on system apps. 
Closing this issue now

closed 2024-11-26 by Frigyes06