Universal-Debloater-Alliance/universal-android-debloater-next-generation #613: removing wallpaper services breaks Lock Star

stepanzak, 2024-09-09:
### Describe the problem

Removing com.samsung.android.dynamiclock (Samsung Wallpaper Services) breaks Lock Star app (part of Good Lock) https://apps.samsung.com/appquery/appDetail.as?appId=com.samsung.systemui.lockstar
While it makes sense, it should still be noted in the package summary.

### You have a solution?

Add this information to the package's summary.

### Provide logs

no

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

schbrns, 2024-09-13:
Correct. 
So does Live Wallpaper Picker.
These are all in my PR.

AnonymousWP, 2024-09-13:
> Correct. So does Live Wallpaper Picker. These are all in my PR.

If you're fixing this issue by a PR, please link it (look at the right side of this issue, 'Development').

AnonymousWP, 2024-10-05:
@schbrns Please follow-up on this.

schbrns, 2024-11-24:
Planned, will close when PR is up.

closed 2024-11-25 by Rudxain