Universal-Debloater-Alliance/universal-android-debloater-next-generation #614: bug(scope): uninstall app in other user

quannguyen-zien, 2024-09-10:
### Describe the bug

i have 2 users when i uninstall Recommended  app of user 0, after that i create user 10. i continue uninstall Recommended  app of user 10, i see it reinstall Recommended  app of user 0.

### Expected behavior

i don't want it reinstall app.

### Operating System

Linux

### Provide logs

2024-09-10 12:19:03 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.angle -> Package com.android.angle installed for user: 0
2024-09-10 12:19:04 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.adservices.api -> Package com.android.adservices.api installed for user: 0
2024-09-10 12:19:04 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.cameraextensions -> Package com.android.cameraextensions installed for user: 0
2024-09-10 12:19:04 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.bluetooth.bthelper -> Package com.android.bluetooth.bthelper installed for user: 0
2024-09-10 12:19:04 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.android.managedprovisioning.overlay -> Success
2024-09-10 12:19:04 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.egg -> Package com.android.egg installed for user: 0
2024-09-10 12:19:04 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.nearby.halfsheet -> Package com.android.nearby.halfsheet installed for user: 0
2024-09-10 12:19:04 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.ondevicepersonalization.services -> Package com.android.ondevicepersonalization.services installed for user: 0
2024-09-10 12:19:04 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.safetycenter.resources -> Package com.android.safetycenter.resources installed for user: 0
2024-09-10 12:19:05 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.theme.icon.square -> Package com.android.theme.icon.square installed for user: 0
2024-09-10 12:19:05 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.theme.icon.pebble -> Package com.android.theme.icon.pebble installed for user: 0
2024-09-10 12:19:05 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.systemui.plugin.globalactions.wallet -> Package com.android.systemui.plugin.globalactions.wallet installed for user: 0
2024-09-10 12:19:06 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 10 com.android.managedprovisioning.overlay -> Success
2024-09-10 12:19:06 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.theme.icon.vessel -> Package com.android.theme.icon.vessel installed for user: 0
2024-09-10 12:19:06 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.theme.icon.taperedrect -> Package com.android.theme.icon.taperedrect installed for user: 0
2024-09-10 12:19:06 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.theme.icon_pack.rounded.systemui -> Package com.android.theme.icon_pack.rounded.systemui installed for user: 0
2024-09-10 12:19:06 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.android.theme.icon_pack.rounded.themepicker -> Package com.android.theme.icon_pack.rounded.themepicker installed for user: 0
2024-09-10 12:19:07 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.ambient.streaming -> Package com.google.ambient.streaming installed for user: 0
2024-09-10 12:19:07 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.accessibility.soundamplifier -> Package com.google.android.accessibility.soundamplifier installed for user: 0
2024-09-10 12:19:07 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.apps.healthdata -> Package com.google.android.apps.healthdata installed for user: 0
2024-09-10 12:19:07 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.apps.googlecamera.fishfood -> Package com.google.android.apps.googlecamera.fishfood installed for user: 0
2024-09-10 12:19:07 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.apps.cbrsnetworkmonitor -> Package com.google.android.apps.cbrsnetworkmonitor installed for user: 0
2024-09-10 12:19:07 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.apps.carrier.log -> Package com.google.android.apps.carrier.log installed for user: 0
2024-09-10 12:19:08 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.apps.internal.betterbug -> Package com.google.android.apps.internal.betterbug installed for user: 0
2024-09-10 12:19:09 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.apps.nbu.files -> Package com.google.android.apps.nbu.files installed for user: 0
2024-09-10 12:19:09 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.apps.recorder -> Package com.google.android.apps.recorder installed for user: 0
2024-09-10 12:19:09 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.apps.restore -> Package com.google.android.apps.restore installed for user: 0
2024-09-10 12:19:09 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.apps.security.securityhub -> Package com.google.android.apps.security.securityhub installed for user: 0
2024-09-10 12:19:09 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.apps.safetyhub -> Package com.google.android.apps.safetyhub installed for user: 0
2024-09-10 12:19:10 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.apps.turbo -> Package com.google.android.apps.turbo installed for user: 0
2024-09-10 12:19:10 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.apps.wearables.maestro.companion -> Package com.google.android.apps.wearables.maestro.companion installed for user: 0
2024-09-10 12:19:10 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.as -> Package com.google.android.as installed for user: 0
2024-09-10 12:19:10 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.as.oss -> Package com.google.android.as.oss installed for user: 0
2024-09-10 12:19:11 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.calendar -> Package com.google.android.calendar installed for user: 0
2024-09-10 12:19:12 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.calculator -> Package com.google.android.calculator installed for user: 0
2024-09-10 12:19:12 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.documentsui.theme.pixel -> Package com.google.android.documentsui.theme.pixel installed for user: 0
2024-09-10 12:19:12 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.feedback -> Package com.google.android.feedback installed for user: 0
2024-09-10 12:19:12 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.gms.location.history -> Package com.google.android.gms.location.history installed for user: 0
2024-09-10 12:19:13 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.markup -> Package com.google.android.markup installed for user: 0
2024-09-10 12:19:13 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.marvin.talkback -> Package com.google.android.marvin.talkback installed for user: 0
2024-09-10 12:19:14 INFO  [src\core\sync.rs:123] [Recommended] cmd package install-existing --user 0 com.google.android.odad -> Package com.google.android.odad installed for user: 0
[... cut ...]
2024-09-10 12:25:54 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.google.android.markup -> Success
2024-09-10 12:25:54 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.google.android.marvin.talkback -> Success
2024-09-10 12:25:54 INFO  [src\core\sync.rs:123] [Recommended] pm uninstall --user 0 com.google.android.odad -> Success

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

schbrns, 2024-09-13:
@quannguyen-zien 
You thought I was joking about being here, huh. Tell those other dudes I dare them to show up.

Rudxain, 2024-12-07:
Android version?

I've noticed that some parts of the code check multi-user support by comparing the Android SDK version to `21`, sometimes with `>` and sometimes `>=`

akha666, 2026-03-25:
Same here
Cross-User Behavior Detected:

Detected cross-user restoration: package exists on user 10 (Enabled) after uninstalling from user 0

This is unusual behavior that may be specific to your device manufacturer (OEM). The package state has been successfully changed on the target user.

The workaround is to uncheck "affect all the users of the device" doesn't help