Universal-Debloater-Alliance/universal-android-debloater-next-generation #622: docs(readme): add `nix develop` command

AnonymousWP, 2024-09-15:
Also if you could add the `nix develop` command in the readme for nix users, that'd be great!

-------- Original Message --------
On 9/15/24 19:53, lavafroth  wrote:

> We can't write tests for a flake. Thanks again for testing it out! 😃
>
> -------- Original Message --------
> On 9/15/24 19:50, Anonymoussaurus  wrote:
>
>> @AnonymousWP approved this pull request.
>>
>> Tested and works. Do we need to write documentation for this before merging or you think it's not necessary?
>>
>> —
>> Reply to this email directly, [view it on GitHub](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/421#pullrequestreview-2305371051), or [unsubscribe](https://github.com/notifications/unsubscribe-auth/AZUKSCD7YZWLKKXNC7LT3DTZWWJT5AVCNFSM6AAAAABGFDGGPOVHI2DSMVQWIX3LMV43YUDVNRWFEZLROVSXG5CSMV3GSZLXHMZDGMBVGM3TCMBVGE).
>> You are receiving this because you were mentioned.Message ID: ***@***.***>

_Originally posted by @lavafroth in https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/421#issuecomment-2351616670_

AnonymousWP, 2025-02-02:
@Rudxain Hey, could you pick this up? Not sure where in the Wiki it would fit.

Rudxain, 2025-02-03:
I'll add it [here](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/Building-from-source)

Rudxain, 2025-02-03:
https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/Building-from-source/b2783d52ca03ee5ed06a318edea550ea087b8f76

I haven't tested if the steps are correct, but the docs seem to imply it's fine

closed 2025-02-03 by Rudxain

h4rldev, 2025-04-30:
Seems to need an update of the flake, the crate needs newer than .76, but the flake with nix develop installs only .76 of rustc and cargo