Universal-Debloater-Alliance/universal-android-debloater-next-generation #623: build(deps): bump dependencies

AnonymousWP, 2024-09-15:
### Describe the feature you want

We need to bump dependencies and check changelogs of some major libraries we use. That way, we can discover whether we can use new IcedRS features for example.

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Marekk2k, 2024-09-15:
My suggestion. Since the export of deleted packages to a txt file has been restored in version 1.1, it would be welcome to import from the discussed file, in order to have greater compatibility in deleting identical packages on other non-debloated Android devices  😉

AnonymousWP, 2024-09-15:
> My suggestion. Since the export of deleted packages to a txt file has been restored in version 1.1, it would be welcome to import from the discussed file, in order to have greater compatibility in deleting identical packages on other non-debloated Android devices 😉

That's a good idea, but not relevant to this issue. Please check if there's an existing issue about it, otherwise create a new one.

PR #949 (2025-07-22): build(deps): upgrade all dependencies but Iced
Closes #623

I've only reviewed a few deps. I've `cargo run` and the app seems to work fine.

Iced-0.13 would require a big rewrite ([IMO, it was overdue](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/discussions/577#discussioncomment-11765767) anyways), but there are bugs (such as #432) and QoL improvements (for users and devs) that would be resolved by it.

We should also update `flake.lock`: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/622#issuec [... cut]
closed 2025-07-22 by Rudxain