Universal-Debloater-Alliance/universal-android-debloater-next-generation #627: bug(theme): `dark-light` v`1.1.1` falls-back to `Light`

Rudxain, 2024-09-16:
### Describe the bug

I've been debugging this and I am 100% certain this is a bug in `dark_light::detect`

### Expected behavior

Theme should be detected properly

### You have a solution?

No. But I'm about to report the issue upstream.

In the meantime, we could downgrade to v`=1.0.0`

### Operating System

Linux

### Distro version

Debian 13

### Window System + Manager

Wayland + Mutter

### Desktop Environment

GNOME

### Provide logs

_Irrelevant_

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

sgse, 2024-09-16:
Firstly, thanks for the app. :-)

As it was so difficult to recognize the logo, I made my own. I didn't know about the 'bug' then. 
If you don't include a .png in the .svg like in the second and third one above, then you also get a clear typeface. It is also much smaller in terms of memory requirements.
Original has 15.1 k
~~~
ls | grep uad-ng
.rw-r--r--  65k root 15 Sep 19:55 uad-ng-1.png
.rw-r--r--  20k root 15 Sep 20:27 uad-ng-1.svg
.rw-r--r-- 4,3k root 15 Sep 20:42 uad-ng.svg
~~~
 
It works for me with the SVG on Arch Based Linux.

[image]

[image]

[image]

Rudxain, 2024-09-16:
@sgse Thank you! but we should talk about it on #629

Frigyes06, 2024-09-19:
Cannot reproduce on Arch Plasma/kwin with Wayland

Frigyes06, 2024-09-19:
Wait I might be dumb, what do you mean by fall back to? Can you provide steps to reproduce?

Rudxain, 2024-09-19:
According to frewsxcv/rust-dark-light#21, the code returns `Light` instead of `Default` when it fails to detect color-scheme. I know it fails on my system, because it always returns `Light` regardless of my GNOME theme

closed 2024-11-03 by adhirajsinghchauhan