Universal-Debloater-Alliance/universal-android-debloater-next-generation #628: feat(scope): Distribute UAD-ng via flatpak

kibashlive, 2024-09-16:
### Describe the feature you want

I'd like to have UAD-ng as a verified package on [flathub](https://flathub.org/).

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-09-16:
#577

I'm unsure if I should close as dupe

AnonymousWP, 2024-09-18:
I prefer having feature requests as issue, not as Discussion.

Rudxain, 2024-09-18:
Should we reopen here? 🤔

I opened the discussion because it's not a feature of the app itself, but rather the format and distribution. And it's unclear which "path" we should choose, although Flatpak seems better than AppImage and Crates.io

AnonymousWP, 2024-09-23:
> Should we reopen here? 🤔
> 
> I opened the discussion because it's not a feature of the app itself, but rather the format and distribution. And it's unclear which "path" we should choose, although Flatpak seems better than AppImage and Crates.io

Do we need code for (automatic) distribution through Flatpak? And if it does, does it need to be in this repo? If not, we can keep it as discussion.

Rudxain, 2024-09-23:
> Do we need code for (automatic) distribution through Flatpak?

I assume yes, it would be part of CI/CD pipeline. In that case it would be called "deployment" rather than "distribution" (I guess?).

More [info here](https://docs.flatpak.org/en/latest/publishing.html)

AnonymousWP, 2024-10-04:
Well yeah, in case it involves code for the pipeline, let's keep this opened indeed.

rhgKc2yp, 2026-11-05:
While waiting for the official one on Flathub, there's https://flatpark.org/apps/io.github.Universal_Debloater_Alliance.uad_ng as well (community maintained)