Universal-Debloater-Alliance/universal-android-debloater-next-generation #631: bug(scope): "Export Uninstalled packages with their description" Success message references .txt file instead of .csv

TMuyi, 2024-09-17:
### Describe the bug

After clicking the export button, the success message mentions that the exported file is a text file. 
However, the exported file in the directory is a .csv file

Looking at the log file, the export operation is not there. For that timetamp, the only operations logged are " ERROR [src\core\utils.rs:99] Can't open the following URL:  "

The exported csv does contain the expected package names & descriptions

### Expected behavior

The pop up shows the actual file extension of the export.

### You have a solution?

Change the success message pop up to reflect the actual file extension

### Operating System

Windows

### Distro version

Win10

### Provide logs

2024-09-17 08:41:26 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2024-09-17 08:41:27 ERROR [src\core\utils.rs:99] Can't open the following URL: 
2024-09-17 08:41:45 ERROR [src\core\utils.rs:99] Can't open the following URL: 
2024-09-17 08:42:07 ERROR [src\core\utils.rs:99] Can't open the following URL: 
2024-09-17 08:43:04 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2024-09-17 08:43:05 ERROR [src\core\utils.rs:99] Can't open the following URL: 

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2025-01-13:
@TMuyi Hi, do you mean this? See https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/796.

TMuyi, 2025-01-14:
Hey @AnonymousWP, yes, that's what I was referring to.

commit fadef4ef (2025-01-14): bug(settings): use csv instead of txt for `uninstalled_packages`

Fixes #631
closed 2025-01-14 by adhirajsinghchauhan