Universal-Debloater-Alliance/universal-android-debloater-next-generation #639: bug(scope): Not working on Windows 10 IoT Enterprise LTSC 2021

RadyanZ, 2024-09-21:
### Describe the bug

It worked just fine on regular windows 10 home. do you guys have any idea about this issue? thanks in advance.

### Expected behavior

That it launches.

### Operating System

Windows

### Distro version

Windows 10 IoT Enterprise LTSC 2021

### Provide logs

Don't have them

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

RadyanZ, 2024-09-21:
I tried the opengl one that was mentioned on the homepage, but the only one I could find was an outdated version, but yes that one worked.

AnonymousWP, 2024-09-21:
Could you please post logs? And if not, please explain why.

RadyanZ, 2024-09-21:
yes, sorry. I'm new to github. 

PS C:\Users\R\Desktop\platform-tools> 2024-09-20 22:21:23 ERROR [C:\Users\runneradmin/.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.19.4\src\vulkan\instance.rs:837] enumerate_adapters: A return array was too small for the result
2024-09-20 22:21:24 ERROR [C:\Users\runneradmin/.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.19.4\src\vulkan\instance.rs:837] enumerate_adapters: A return array was too small for the result

RadyanZ, 2024-09-21:
Nevermind, I fixed it with a new amd driver. Thanks for your time.

closed 2024-09-21 by AnonymousWP