Universal-Debloater-Alliance/universal-android-debloater-next-generation #640: Recommended List suggestions are counterproductive and needs additional individual review

ambhaiji, 2024-09-21:
### Describe the problem

Before proceeding further after writing most of this I do realize that some of this is carried over from the original app this is based off.

after uninstalling the entire recommended list and having several problem I have come to the conclusion that the recommended list is made in a way where it will recommend you to uninstall everything to the point just enough to not crash your system. but breaks alot of functionality along the way, several functions I consider basic and some fairly alarming removals I have noticed trying undo majority of the installations.

It breaks any kind of device to device discovery, Whatsapp transfer just couldn't find the debloated phone.
Quick Share can't find the debloated device.
Samsung Find My Mobile is part of the recommended uninstalls which I find bonkers.

com.samsung.android.app.telephonyui reads "I don't know how important this app is." 
I uninstalled it and it didn't brick my phone so it must be okay mentality

com.google.android.gms.supervision
"it has something to Family Link parental controls"
bad luck if you have kids I guess

idk, i guess, basically, something are part of the description of so many apps making just a bunch of yolo uninstalls.

The recommended list is basically useless if I have to take the time and study all the now effectively suggestions on it one by one to see what I want to remove.

the idea of the recommended list should not be to instantly make it feel like AOSP, but to remove Bloat.

### You have a solution?

first problem to fix is allow searching the description, so we can search by what an app does instead of knowing its secret magic abbreviation, semi-abbreviation or weird codename that any particular apps decided to use.
nearby device scanning supposedly works with com.samsung.android.easysetup and com.samsung.android.beacon.manager none of where the app names itself have any relevance indicating its sharing or networking capabilities.

remove the recommended list, replace it with "popular bloatware', stuff like Facebook, Samsung Free etc

removing any app name containing 'service' should probably be moved to Advanced as it probably breaks some service.

Apps that concern privacy should be listed separately. 

screen and audio mirroring stuff, things normal people do should not be recommended removals and also be moved to Advanced, clearly indicating that the phone will continue to work if removed but missing specific functionality.

### Provide logs

A few honorable crazy descriptions 

------------------------------------------------------
com.sec.location.nfwlocationprivacy

Service provider location

Not needed for location, | found something like that:

{This app enables your carrier to access your location for network improvement **or other reasons**)
**This app is hidden so idk how useful it is.**
------------------------------------------------------
com.sec.modem.settings

This package runs at startup and logs things (related to the modem ?). Seems Pretty shady to me (I don't like its orwellian name).

WAT?
------------------------------------------------------

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-09-22:
Essentially dupe of #583. But, thank you for elaborating!

About searching descriptions, I assumed we already had an Issue about it. Turns out, it [only exists in the OG repo](https://github.com/0x192/universal-android-debloater/issues/134)

Rudxain, 2024-09-23:
I've opened #647 for better organization, as this issue requests 2 things

closed 2024-09-23 by Rudxain