Universal-Debloater-Alliance/universal-android-debloater-next-generation #642: feat(GUI): remove pack-state buttons

Rudxain, 2024-09-22:
### Describe the feature you want

This would simplify the code (less maintenance burden) and it would make the app safer (users are less likely to uninstall on accident)

I believe the "Review selection" button should be the primary way to change the state of 1 or more packages, as it serves as a confirmation-dialog before doing a potentially-irreversible action (`pm clear`)

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-09-22:
BTW, everytime I scroll through the pack-list, I feel anxious, because the mouse is dangerously close to the buttons

AnonymousWP, 2024-09-23:
> BTW, everytime I scroll through the pack-list, I feel anxious, because the mouse is dangerously close to the buttons

Maybe we can make confirmation message boxes for the `Uninstall` and `Disable` buttons.

Rudxain, 2024-09-23:
I forgot to mention the other problem: When filtering packs to show "Enabled" only, uninstalling/disabling a pack will make it silently "dissappear" from the (filtered) list, leaving no trace of the action (unless looking at the log). Same thing happens when looking at uninstalled/disabled packs, then restoring 1 or more.

A way to improve this, is to always keep the pack in the current view until the user changes any of the filters