Universal-Debloater-Alliance/universal-android-debloater-next-generation #644: bug(scope): Does not work on Huawei MediaPad T1 7.0

FahadBinHussain, 2024-09-23:
### Describe the bug

When I click Uninstall beside Chrome, It shows uninstalled, but when I refresh the list, chrome appears again as installed, also on the device, Chrome still works as usual.

### Expected behavior

Uninstall packages like other phones

### You have a solution?

No

### Operating System

Windows

### Distro version

Win11 Pro

### Provide logs

[UAD_20240923.log](https://github.com/user-attachments/files/17095398/UAD_20240923.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-09-23:
As you can see in the logs, ADB can't be found. Did you add ADB to PATH?

FahadBinHussain, 2024-10-01:
> As you can see in the logs, ADB can't be found. Did you add ADB to PATH?

Yes, I added ADB later to path, and tried again. Still didn't work

AnonymousWP, 2024-10-04:
Looks like we're having another issue here: as you can see in your logs, your MediaPad runs Android 4.4, which can only use the `block` method: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ#android-44-kitkat. It means Chrome is now blocked from being run, i.e. disabled. You can't uninstall it with UAD-ng, maybe manually? Not sure about that old Android version.

FahadBinHussain, 2024-10-13:
> Looks like we're having another issue here: as you can see in your logs, your MediaPad runs Android 4.4, which can only use the `block` method: [...]

Chrome is still working as usual, Not Disabled or Uninstalled.

AnonymousWP, 2024-10-18:
Assuming you tried to execute the command manually as well, it's either a bug in ADB, a bug in your Android skin or (most likely) your OEM doesn't allow you to uninstall/block Chrome.

closed 2024-10-18 by AnonymousWP

FahadBinHussain, 2024-10-21:
> Assuming you tried to execute the command manually as well, it's either a bug in ADB, a bug in your Android skin or (most likely) your OEM doesn't allow you to uninstall/block Chrome.

No, I didn't try manually executing any adb command, I'll try and update.