Universal-Debloater-Alliance/universal-android-debloater-next-generation #645: build(ci): provide non-`wgpu` builds

AnonymousWP, 2024-09-23:
> builds with and without `wgpu` if that's possible?

We should try:
1. create a branch that disables `wgpu`
2. generate build-artifacts from the branch
3. ask users to try running those artifacts

_Originally posted by @Rudxain in https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/599#issuecomment-2352085265_

Rudxain, 2025-01-09:
Wait, I just realized something. We have to provide `tiny_skia` builds. `wgpu` can't be ditched

AnonymousWP, 2025-01-12:
> Wait, I just realized something. We have to provide `tiny_skia` builds. `wgpu` can't be ditched

Not sure how I need to do that to be honest, not familiar with it. Do you know? Then I can implement it.

AnonymousWP, 2025-01-12:
Would this work?

```yml
name: Build artifacts
on:
  workflow_dispatch:
  workflow_call:

jobs:
  build:
    name: Building ${{ matrix.build_target }} [${{ matrix.update_feature }}]
    runs-on: ${{ matrix.os }}
    strategy:
      matrix:
        build_target: [linux, macos, macos-13, windows]
        update_feature: [self-update, no-self-update]
        exclude:
          - build_target: windows
            update_feature: no-self-update
          - build_target: macos-13
            os: macos-latest # Avoid duplicate runs on macos-latest
        include:
          - build_target: linux
            os: ubuntu-latest
          - build_target: macos
            os: macos-latest
          - build_target: macos-13
            os: macos-13
          - build_target: windows
            os: windows-latest
          - update_feature: self-update
            update_name: "" # we don't want this in the binary filename
          - update_feature: no-self-update
            update_name: "-noselfupdate"
    steps:
      - uses: actions/checkout@v4
      - uses: dtolnay/rust-toolchain@stable
      - uses: rui314/setup-mold@v1 # faster linker
      - uses: actions/cache@v4
        with:
          path: |
            ~/.cargo/bin/
            ~/.cargo/registry/index/
            ~/.cargo/registry/cache/
            ~/.cargo/git/db/
            target
          key: ${{ runner.os }}-release-${{ hashFiles('**/Cargo.lock') }}
          restore-keys: ${{ runner.OS }}-release-
        if: matrix.os == 'ubuntu-latest'
      - name: Building
        run: cargo build --release --no-default-features --features wgpu,tiny_skia,${{ matrix.update_feature }}
      - name: Renaming binaries [Windows]
        if: matrix.os == 'windows-latest'
        run: mv target/release/uad-ng.exe uad-ng-${{ matrix.build_target }}.exe
      - name: Renaming binaries [Others]
        if: matrix.os != 'windows-latest'
        run: mv target/release/uad-ng uad-ng${{ matrix.update_name }}-${{ matrix.build_target }}
      - name: Tarball Linux/MacOS binary
        if: matrix.os != 'windows-latest'
        run: tar -czf uad-ng${{ matrix.update_name }}-${{ matrix.build_target }}{.tar.gz,}
      - name: Install coreutils for macOS
        if: startsWith(matrix.os, 'macos')
        run: brew install coreutils
      - name: Create checksums for binaries and archives [Windows]
        if: matrix.os == 'windows-latest'
        run: sha256sum uad-ng-${{ matrix.build_target }}.exe | tee uad-ng-${{ matrix.build_target }}.exe-checksum.txt
      - name: Create checksums for binaries and archives [Others]
        if: matrix.os != 'windows-latest'
        run: |
          sha256sum uad-ng${{ matrix.update_name }}-${{ matrix.build_target }} | tee uad-ng${{ matrix.update_name }}-${{ matrix.build_target }}-checksum
          sha256sum uad-ng${{ matrix.update_name }}-${{ matrix.build_target }}.tar.gz | tee uad-ng${{ matrix.update_name }}-${{ matrix.build_target }}.tar.gz-checksum
      - name: Upload artifacts
        uses: actions/upload-artifact@v4
        with:
          name: uad-ng${{ matrix.update_name }}-${{ matrix.build_target }}
          path: uad-ng-*
```

Rudxain, 2025-01-12:
> Not sure how I need to do that to be honest, not familiar with it. Do you know? Then I can implement it.

I expect it would be like going back in time, when there were OpenGL builds. A `git bisect` would be helpful

> Would this work?
>
> [...]

I'm unsure, TBH :sweat_smile:

> I enabled both `tiny_skia` and `wgpu` as features.

I guess that would work

AnonymousWP, 2025-02-17:
No longer relevant due to https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/850.

closed 2025-02-17 by AnonymousWP