Universal-Debloater-Alliance/universal-android-debloater-next-generation #649: Some apps are incorrectly labeled as recommended.

Unknown-Indo, 2024-09-24:
### List your packages

- Package name: com.google.android.overlay.gmsconfig.common & com.google.android.overlay.modules.modulemetadata.forframework.
- Current list: All of the mentioned packages are google.
  - Proposed list: Same as current list.
- Current removal: All of the mentioned packages are recommended.
  - Proposed removal: com.google.android.overlay.gmsconfig.common must be expert and com.google.android.overlay.modules.modulemetadata.forframework must be unsafe.
- Current description:

com.google.android.overlay.gmsconfig.common: Useless configurations about webview, location, lists google apps. Everything works without it.

com.google.android.overlay.modules.modulemetadata.forframework: config_defaultModuleMetadataProvider
It looks like useless module metadata from google
Not needed, it doesnt affect on (com.google.android.modulemetadata)

  - Proposed description: 

com.google.android.overlay.gmsconfig.common: If you delete this package, you won't be able to log in to your google account on a google app. But if you don't need to log in to a google account, you can safely delete this.

com.google.android.overlay.modules.modulemetadata.forframework: If you delete this package, your has an extremely large chance of bootlooping. Just don't try messing around with this package by deleting it. Tested on a Samsung A34 5G.

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-09-29:
I can tell by the names alone that those should be (at least) `Advanced`! I agree with the `removal` change

Rudxain, 2024-09-29:
On SM-A315G, none of those can be disabled via Settings app, but battery-use can be `Restricted`

Unknown-Indo, 2024-09-29:
> I can tell by the names alone that those should be (at least) `Advanced`! I agree with the `removal` change

👍

AnonymousWP, 2024-10-04:
I'll try working on it this weekend, I'm pretty busy and currently the only maintainer for packages.

Unknown-Indo, 2024-10-05:
> I'll try working on it this weekend, I'm pretty busy and currently the only maintainer for packages.

👍

Samscryption, 2024-10-09:
Deleting **com.google.android.overlay.gmsconfig.common** caused a few of my apps to constantly crash even when they had nothing to do with Google or Google accounts so this definitely should be in advanced.

Rudxain, 2024-10-09:
> nothing to do with Google

Some apps use the Google Play Services APIs under the hood. I remember the Discord app failed to fetch any data if GMS/GPS was disabled, despite the web and Electron versions working perfectly fine without GMS.

WhatsApp also makes use of GMS to get messages in real-time (only when the app is in the background).

GMS is one of Google's "tentacles" over Android. [This is not even a conspiracy theory](https://arstechnica.com/gadgets/2018/07/googles-iron-grip-on-android-controlling-open-source-by-any-means-necessary/), sadly

Rudxain, 2024-10-09:
Thanks for the info anyways! It does help the documentation

closed 2024-10-10 by AnonymousWP

Samscryption, 2024-10-10:
> Some apps use the Google Play Services APIs under the hood. I remember the Discord app failed to fetch any data if GMS/GPS was disabled, despite the web and Electron versions working perfectly fine [...]

I wonder if I can use revanced/microg gms core to replace the functionality of this app? 

https://github.com/ReVanced/GmsCore

Rudxain, 2024-10-11:
> I wonder if I can use revanced/microg gms core to replace the functionality of this app?

Yes, with some caveats:
- [sign-spoof required](https://github.com/microg/GmsCore/wiki/Signature-Spoofing)
- may need to unlock boot-loader
- missing G Play System Updates (Citation needed. This is an assumption)