Universal-Debloater-Alliance/universal-android-debloater-next-generation #650: bug(perf): auto-theme introduces performance regression

Rudxain, 2024-09-24:
### Describe the bug

#531, more specifically this commit 76aba1e54829dd5fef2f91294f837dbcfa28e3e4, causes the **entire UI** to become sluggish on a Dell Inspiron 15R.

IDK exact perf, but I guess previous commit is ~60FPS and the one I referenced is ~1FPS (no device connected) or **~0.5FPS** (when the list of installed packs has been fetched)

### Expected behavior

UAD-NG must have smooth performance while idle

### You have a solution?

I suspect Iced is somehow calling `dark_light::detect` **every frame** 💀 (possibly our fault).

If we upgrade to [the new Iced](https://github.com/iced-rs/iced/releases/tag/0.13.0), then the `dark_light` integration should work properly as-is

### Operating System

Linux

### Distro version

Debian 13

### Window System + Manager

Wayland + Mutter

### Desktop Environment

GNOME 46 (some apps are v47)

### Provide logs

This may be irrelevant:
```
2024-09-24 13:34:41 WARN  [/home/rudxain/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/instance.rs:696] InstanceFlags::VALIDATION requested, but unable to find layer: VK_LAYER_KHRONOS_validation
MESA-INTEL: warning: Ivy Bridge Vulkan support is incomplete
2024-09-24 13:34:41 WARN  [/home/rudxain/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/adapter.rs:1143] Adapter is not Vulkan compliant, hiding adapter: Intel(R) HD Graphics 4000 (IVB GT2)
```

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

neingeist, 2024-09-25:
This makes the UI _very_ slow, btw!

Rudxain, 2024-09-25:
I've confirmed that if the user selects any theme other than "Auto", the UI performance is fast.

Being optimistic, this means there's a workaround for users who don't want to await the next release (minor or hotfix patch). The feature is broken anyways (#627), so I doubt most people would use it as-is 🥲

Rudxain, 2024-10-26:
> I doubt most people would use it

I was wrong. It's the **default** setting:
https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/c3fe1a466615406267847650391cf885a2246768/src/core/theme.rs#L7-L9

We should fix this ASAP

AnonymousWP, 2024-10-27:
> > I doubt most people would use it
> 
> I was wrong. It's the **default** setting:
> 
> https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/c3fe1a466615406267847650391cf885a2246768/src/core/theme.rs#L7-L9
> 
> We should fix this ASAP

Is it an easy fix?

Rudxain, 2024-10-27:
> Is it an easy fix?

Yes, we can simply move the attribute to `Dark` (as per #540)!

But the config file on some users' devices will still be set to "Auto". So UADNG should deem `Auto` as invalid (as if the theme didn't exist) and fall-back to `Dark`

Rudxain, 2024-10-28:
> I suspect Iced is somehow calling `dark_light::detect`

[Found the culprit](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/47980f635eb4f4ae18dd57eee6c001bf4a7d2b0e/src/gui/style.rs) (search `palette()` to see the horror 💀)

Rudxain, 2024-10-28:
I'm already working on using `static_init` to cache `dark_light` on startup. This should fix the perf problem and patch (not a real fix) #550 !

AnonymousWP, 2024-10-28:
Great work!

closed 2024-11-03 by adhirajsinghchauhan

Rudxain, 2025-02-28:
If you're using the [stable release](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/releases/tag/v1.1.0), the issue [hasn't been fixed there](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/Home/4e96051b39836dcff8af8b46b16866c045ae46d0#features). However, this bug has been fixed on the `main` branch.

We're considering doing a patch release before [the milestone](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/milestone/6). But before that, I have to revert a change I made, because I realized it might introduce a bug when handling multi-user Androids. I apologize for the inconvenience 😦