Universal-Debloater-Alliance/universal-android-debloater-next-generation #657: feat(scope): Action History Functionality

kat0d, 2024-09-27:
### Describe the feature you want

Please, add action history, like, when I uninstall/disable package, or restore/enable.
Because there is a situation. Like, I had remove some uninstall packages from recomended on Xiaomi android 14. And my mobile internet stop working. It was impossible to add APN Point ("Access Point Name settings are not available for this user"). I had enable/restore few packages, and it's all fine.
And the point is, that I can't remember, what packages I had restore. And if there was a history, then I could determine which package should not be removed

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-09-29:
> ### Describe the feature you want
> Please, add action history, like, when I uninstall/disable package, or restore/enable. Because there is a situation. Like, I had remove some uninstall packages from [...]

There are logs for that: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ#where-are-the-logfiles

Rudxain, 2024-09-29:
I agree that the logs should be enough, as they can be searched/`grep`ed. Unless they meant something like _"undo/redo"._

Some file managers support undoing reversible actions, such as moving/renaming files, but not undoing a deletion (unless the files were sent to the recycle-bin)