Universal-Debloater-Alliance/universal-android-debloater-next-generation #658: bug(scope): Apps supposedly listed as recommanded don't appear in the list (tiktok, facebook...)

vaccarov, 2024-09-27:
### Describe the bug

Hello all,

First let me thank all the community for this tool and for the maintenance.

I'd like to submit what i think is a bug, indeed after full debloat of recommended apps, I still have apps that [should have been listed](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/main/resources/assets/uad_lists.json) but don't show in the GUI, just a couple of examples:

```
tiktok: com.zhiliaoapp.musically
amazon music: com.amazon.mp3
booking: com.booking
...
```

Also, some obvious bloatware do not appear in uad_lists.json, for example here is one:
```
aliexpress: com.alibaba.aliexpresshd
```

My device is a Poco M6 pro (2312fpca6g), last update (hyperos 1.0.5.0 UNFMIXM)

Let me know if anything else is needed, I'd be happy to help !

### Expected behavior

Apps listed in uad_lists should appear in the GUI

### You have a solution?

Manual uninstallation from phone

### Operating System

MacOS

### Distro version

14.6.1

### Provide logs

<details><summary>Log</summary>
```
2024-09-27 14:24:50 ERROR [src/core/config.rs:92] Invalid config file: `TOML parse error at line 1, column 1
  |
1 | [general]
  | ^^^^^^^^^
missing field `backup_folder`
`
2024-09-27 14:24:50 ERROR [src/core/config.rs:96] Restoring default config file
2024-09-27 14:24:51 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
2024-09-27 14:24:52 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2024-09-27 14:24:52 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 33 | DEVICE: POCO 2312FPCA6G
2024-09-27 14:24:52 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------
2024-09-27 14:24:59 DEBUG [src/core/update.rs:204] Failed to check UAD-ng update
2024-09-27 14:25:37 DEBUG [src/core/config.rs:80] config: New device settings saved
2024-09-27 14:25:45 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
2024-09-27 14:25:47 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2024-09-27 14:25:47 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 33 | DEVICE: POCO 2312FPCA6G
2024-09-27 14:25:47 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------
2024-09-27 14:25:53 DEBUG [src/core/update.rs:204] Failed to check UAD-ng update
2024-09-27 14:27:53 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
2024-09-27 14:27:53 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2024-09-27 14:27:53 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 33 | DEVICE: POCO 2312FPCA6G
2024-09-27 14:27:53 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------
2024-09-27 14:28:01 DEBUG [src/core/update.rs:204] Failed to check UAD-ng update
2024-09-27 14:57:14 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2024-09-27 14:57:14 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 34 | DEVICE: POCO 2312FPCA6G
2024-09-27 14:57:14 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------
[... 129 lines cut ...]
2024-09-27 15:17:39 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.xiaomi.xmsfkeeper -> Success
2024-09-27 15:17:39 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 org.ifaa.aidl.manager -> Success
2024-09-27 15:17:56 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2024-09-27 15:17:56 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 34 | DEVICE: POCO 2312FPCA6G
2024-09-27 15:17:56 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------
2024-09-27 15:28:33 INFO  [src/core/sync.rs:131] [Advanced] pm uninstall --user 0 com.android.sharedstoragebackup -> Success
2024-09-27 15:28:33 INFO  [src/core/sync.rs:131] [Advanced] pm uninstall --user 0 com.miui.mediaviewer -> Success
2024-09-27 15:28:34 INFO  [src/core/sync.rs:131] [Advanced] pm uninstall --user 0 com.miui.gallery -> Success
2024-09-27 15:28:34 INFO  [src/core/sync.rs:131] [Advanced] pm uninstall --user 0 com.google.android.partnersetup -> Success
2024-09-27 15:28:34 INFO  [src/core/sync.rs:131] [Advanced] pm uninstall --user 0 com.miui.global.packageinstaller -> Success
2024-09-27 15:28:34 INFO  [src/core/sync.rs:131] [Advanced] pm uninstall --user 0 com.android.providers.downloads.ui -> Success
2024-09-27 15:28:34 INFO  [src/core/sync.rs:131] [Advanced] pm uninstall --user 0 com.google.android.printservice.recommendation -> Success
2024-09-27 15:28:34 INFO  [src/core/sync.rs:131] [Advanced] pm uninstall --user 0 android.miui.poco.launcher.res -> Success
2024-09-27 15:28:34 INFO  [src/core/sync.rs:131] [Advanced] pm uninstall --user 0 com.xiaomi.account -> Success
2024-09-27 15:28:49 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2024-09-27 15:28:49 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 34 | DEVICE: POCO 2312FPCA6G
2024-09-27 15:28:49 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------
```
</details>

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-09-29:
https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/592#issuecomment-2352138951

Rudxain, 2024-09-29:
I'm closing this as dupe. But thank you anyways!

If anyone believes this isn't a dupe of #592, please let me know

closed 2024-09-29 by Rudxain