Universal-Debloater-Alliance/universal-android-debloater-next-generation #662: bug(scope): Very slow and laggy UI experience on kali linux .

StarMusa88, 2024-10-02:
### Describe the bug

Very slow and laggy UI experience compared to the original UAD on kali linux .But ubuntu and windows 11 seems fine. just wondering what could be the issue ??.

### Expected behavior

N/A

### You have a solution?

N/A

### Operating System

Linux

### Distro version

Kali Linux, Win 11 , Ubuntu

### Window System + Manager

xorg

### Desktop Environment

Gnome

### Provide logs

<details><summary>Log</summary>

```
Please replace this line by the log-file contents
```
</details>

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-10-04:
Which version of UAD-ng?

StarMusa88, 2024-10-04:
Hi,

version1.1.0

theme=Automatic

https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/releases/download/v1.1.0/uad-ng-linux

closed 2024-10-04 by StarMusa88

AnonymousWP, 2024-10-05:
Yes it's a known issue.

Rudxain, 2024-11-06:
It'll be fixed in the next release. See #650