Universal-Debloater-Alliance/universal-android-debloater-next-generation #670: feat(scope): We should consider disabling apps IN ADDITION to uninstalling them

celenityy, 2024-10-05:
### Describe the feature you want

Hi there,

Thank you for your great work on this project.

I believe that when uninstalling an app through UAD, rather than only uninstalling it, we should uninstall **and** disable the app. 

*(Yes, it is possible to do both... Android is weird sometimes :p).*

___ 

The reason I think this would be worth pursuing is that I have noticed instances where uninstalling apps alone causes breakage, as other apps on the device think the app is still installed, so they will still try to call the app as a dependency. But of course... the app doesn't actually exist since we uninstalled it. The end result causes unnecessary issues & breakage. **When an app is both disabled & uninstalled, this is no longer a problem, as the app can see the dependent app is unavailable and fallback appropriately.**

For examples:

1. I've noticed on Android TV devices, Google's built-in screensaver completely breaks & refuses to launch if Play Services is uninstalled. **If Play Services is both disabled & uninstalled, the issue resolves.**
2. I've noticed some apps such as Netflix will complain & give errors that Google Play Services is "updating or temporarily unavailable." **Again, once Play Services is disabled in addition the uninstallation, the error goes away, and the app works as expected.**
3. Similar issues can be commonly seen with Signal, where the app completely breaks if Play Services is uninstalled, but not disabled, giving the same error as Netflix above. Ex. https://github.com/signalapp/Signal-Android/issues/10368 https://github.com/signalapp/Signal-Android/issues/11139 https://github.com/signalapp/Signal-Android/issues/9279 - **As with the other 2 examples, Signal works perfectly fine once Play Services is also disabled, as Signal can then use its fallback code.**

All 3 of these examples are with Play Services - but I imagine there are many other cases like this, especially with different system apps. In fact, I honestly wouldn't be surprised if this causes a majority of the issues people have when debloating & removing certain apps from their device. I believe that this if this change is implemented, it would fix a **lot** of problems.

___ 

To sum up my proposal:

* Leave the `disable` feature as is. 
* Change `uninstall` to both uninstall **and** disable apps, rather than **only** uninstall it.
* Similar when reinstalling apps removed this way, instead of only re-installing the app, it would also re-install **and** re-enable it.

I'm curious to hear thoughts on this - I'm not aware of any downsides this would introduce, but I'm curious if there's something I'm missing?

🙏

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-10-23:
Maybe I'm missing something here, but I don't entirely understand: if you uninstall, and it succeeds, then how you want to disable an app? After all, the app is gone from the device, so it can't be uninstalled. Even if you disable first, then uninstall, other apps will still see it's no longer present.

In my opinion a better feature is to detect whether uninstalling fails. If so - for example due to the lack of permissions - automatically try disabling instead.

celenityy, 2024-10-23:
@AnonymousWP Thanks for your response.

> if you uninstall, and it succeeds, then how you want to disable an app? After all, the app is gone from the device, so it can't be uninstalled.

Android allows both uninstalling **and** disabling apps with ADB. Yes, it's weird, and it doesn't make much sense logically (I suspect its something to do with the fact that the app is still available on the system partition, but uninstalled on the user level). You can test this out and see for yourself.

The problem (as demonstrated by my examples above) is that when apps are only uninstalled via ADB (& **not** also disabled), the other apps on the device can't see that they're uninstalled, and may try calling them as a dependency or communicate for certain functionality. (Again, I think this is probably related to the fact that the app is still on the system partition even though we uninstalled it for the user, so apps on the device still think its available, but when they try to call it... of course nothing is there, it isn't present, so it can cause & causes issues, as I highlighted in my examples above). 

**However**, apps on the device are able to detect when an app is disabled, meaning rather than trying to call the uninstalled app that doesn't exist, when it's *also* disabled, the app can fallback appropriately & won't crash or display error messages.

>  if you uninstall, and it succeeds, then how you want to disable an app?

Simple. (Replace `com.package.example` below with any app of your choice):

`adb shell pm uninstall --user 0 com.package.example && adb shell pm disable-user --user 0 com.package.example`

Like I said above, feel free to test this yourself, and I think you'll see what I mean.

Rudxain, 2024-10-24:
I'll test the "hack" when I have some free time.

> detect whether uninstalling fails. If so automatically try disabling instead.

#632 (I'm mentioning it to link the issues together). If both issues are "closed as completed" then this implies there would be no need for fail-detection, as the pack will already be disabled by the time UADNG notices the error (unless the `disable` itself fails, which it should notify to the user).

This might be order-sensitive and device-dependent, so we have to ask users to test both `disable`->`uninstall` and `uninstall`->`disable` and report their findings 👀

AnonymousWP, 2024-10-29:
Thanks for the thorough explanation both of you.🙂It sounds interesting, but I'm afraid we won't (and can't, due to members' free time, or rather lack of) work on it anytime soon.