Universal-Debloater-Alliance/universal-android-debloater-next-generation #691: pkg(scope): Battery Drainage

ItsEagle, 2024-10-14:
### Describe the problem

hi, ever since i debloated my phone ive noticed the battery seems to drain faster than with all the bloat on which is pretty weird. I was wondering if you heard of similar cases of this happening, all i could find was one guy on reddit addressing such an issue but no proper replies were formed. Im using a Samsung Galaxy A71 with Android 11 on it 

### Provide logs

󠀀󠀀󠀀󠀀󠀀󠀀Not available at this moment

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Blademind, 2024-10-16:
Can relate

AnonymousWP, 2024-10-16:
Then you probably removed a package that's trying to be called by another package. Post logs or your uninstalled logs first. Or try restoring and see if it's solved.

ItsEagle, 2024-10-16:
> Then you probably removed a package that's trying to be called by another package. Post logs or your uninstalled logs first. Or try restoring and see if it's solved.

i've uploaded the log file, sorry for the inconvenience before

AnonymousWP, 2024-10-18:
I have no idea which of the many packages you installed will cause battery drain. You'll have to find out for yourself by:

1. Re-install all apps
2. Uninstall on a per-batch basis (per 5, per 10)

We can't solve this for you, so I'm closing this.

closed 2024-10-18 by AnonymousWP