Universal-Debloater-Alliance/universal-android-debloater-next-generation #694: pkg(scope): verizon boot up and notification still showing

aerobool90, 2024-10-18:
### Describe the problem

so even after uninstalling the verizon carrier apps with UAD i still see verizon boot up and notification. my phone is unlocked and works with any carrier its just that its stuck with these verizon things popping up.

### Provide logs

[UAD_20241018.log](https://github.com/user-attachments/files/17431722/UAD_20241018.log)
[uad_lists.json](https://github.com/user-attachments/files/17431724/uad_lists.json)

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

aerobool90, 2024-11-10:
> Seems like you often tried it with user 150, which is not allowed. Try to uninstall those apps with user 0.

whats user150

AnonymousWP, 2024-11-10:
> > Seems like you often tried it with user 150, which is not allowed. Try to uninstall those apps with user 0.
> 
> whats user150

Probably a work profile.

aerobool90, 2024-11-20:
> > > Seems like you often tried it with user 150, which is not allowed. Try to uninstall those apps with user 0.
> > 
> > 
> > whats user150
> 
> Probably a work profile.

idk what a work profile is. how do i uninstall with user 0? im so lost

schbrns, 2024-11-25:
150 is Secure Folder.
This is not a UAD issue, you should reach out to your carrier.
Even if Verizon weren't your carrier, this would be a firmware issue.

Please feel free to re-open the issue if you have any questions.

closed 2024-11-25 by schbrns

aerobool90, 2024-12-01:
hello, I need guidance on using UAD's "user 0" function as it was recommended in the resolution.
 this functionality is integral to UAD, so their expertise is needed.

Rudxain, 2024-12-02:
[Disabling "Setup Wizard" on a Galaxy J3 Mission blocks all the annoying "Non-Verizon SIM" notifications](https://xdaforums.com/t/q-how-to-bypass-or-get-rid-of-activation-agent.2921569/)