Universal-Debloater-Alliance/universal-android-debloater-next-generation #697: Update description (com.samsung.SMT)

60063, 2024-10-23:
### List your packages

Note in the description that deleting or disabling com.samsung.SMT causes not loading correctly "Language Packs" in "General Management" settings on Samsung (languages ​​are not loading)

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-10-26:
Do you mean Samsung Keyboard languages, or the OS lang-list?

skaeu, 2024-10-26:
> Do you mean Samsung Keyboard languages, or the OS lang-list?

This section is not related to OS languages ​​and Samsung keyboard, the Language Packs section is related to, for example, AI functions and so on.

closed 2024-11-25 by AnonymousWP