Universal-Debloater-Alliance/universal-android-debloater-next-generation #698: pkg(oem): Update com.xiaomi.bluetooth as safe to remove

Akuma73-sudo, 2024-10-24:
### List your packages

Package: **com.xiaomi.bluetooth**

- Current list: Expert
  - Proposed: Advanced
  
  
- Current desc: 
MIUI Bluetooth Control
You must keep this if you want bluetooth to work properly.
  - Proposed: 
  MIUI Bluetooth Extension
  Doesn't seem to affect bluetooth functionality. Tested on Poco M4 Pro 5G HyperOS 1.0.1.0 
  Note: If you deleted (com.xiaomi.xmsf) this package will likely send "Bluetooth extension stopped working" errors, uninstalling it removes them.

 1: Uninstalling Xiaomi stuff + "**com.xiaomi.xmsf**" causes "**com.xiaomi.bluetooth**" to make errors [reddit](https://www.reddit.com/r/XiaomiGlobal/comments/1ff3bs6/bluetooth_extension_error_xiaomi). Bluetooth works fine as **com.android.bluetooth** handles connections ([source](https://www.reddit.com/r/XiaomiGlobal/comments/1ff3bs6/comment/lpv8rnn)). In my case, uninstalling "**com.xiaomi.bluetooth**" also doesn't seem to affect bluetooth functionality. Tested working without this package: TWS headphones, phone file sharing.

2: [Alternative temporary fix](https://xdaforums.com/t/bluetooth-extension-error.4652662/):  clearing data of "Bluetooth Extension" app removes errors, but they return after restart.

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

schbrns, 2024-11-25:
@Akuma73-sudo 
https://blog.oversecured.com/20-Security-Issues-Found-in-Xiaomi-Devices/
Do you mind creating a PR with all of these changes? 
Please, and thank you.

Akuma73-sudo, 2024-11-26:
@schbrns 
I don't know how to. So far I had no issues with using anything related to Bluetooth with these uninstalled packages: [uninstalled_packages_20241126.zip](https://github.com/user-attachments/files/17919588/uninstalled_packages_20241126.zip)
I hope this helps.

schbrns, 2024-12-26:
I'll get the one change done but we should maybe create a new issue with those outstanding packages from the link and/or any from the zip provided above. Thanks for providing that.

closed 2025-02-05 by AnonymousWP