Universal-Debloater-Alliance/universal-android-debloater-next-generation #703: bug(adb): "Adb not installed" error even though the reboot worked

GingrPowr, 2024-10-28:
### Describe the bug

I think that since I didn't installed ADB the way this project wants to (I used it before), after clicking [Reboot] the "ADB not installed on your system" error is displayed. But my connected device still rebooted, because I added my ADB install folder into my PATH environment variables. Worth to mention: it also worked before, to uninstall the 120+ bloatware that were reinstalled on my freshely reseted Samsung J5. So ADB is installed, works (I used it before today), is detected by the app and works with the app.

### Expected behavior

Either:
- Fail and display error message,
- Work and don't display error message.

### You have a solution?

Either:
- Do not use ADB installed in a different manner than what is asked in your installation guideline -> which would be a shame since it's currently working,
- Update the installation guideline, and the app error check, and keep using the ADB found in PATH -> preferred option.

### Operating System

Windows

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

VideoBeeYT, 2024-10-31:
I have same issue, it's such a hassle that they didn't think about the simplest thing ever

zicke2, 2024-11-12:
Download android platform tools and unzip it somewhere.
https://dl.google.com/android/repository/platform-tools-latest-windows.zip

Put the tool into this Folder and start it from there… it will work ;)

greets from austria
[image]

GingrPowr, 2024-11-13:
> Download android platform tools and unzip it somewhere. https://dl.google.com/android/repository/platform-tools-latest-windows.zip
> 
> Put the tool into this Folder and start it from there… it will work ;)
No, that will not resolve the bug, only hide it.

Frigyes06, 2024-11-26:
#700 related

closed 2024-11-26 by Frigyes06

GingrPowr, 2024-11-26:
@Frigyes06 This is not a feature or an enhancement.

This is a bug to be fixed: the software is stating something and doing the inverse, literally working while it says it won't.
Also I don't see how this bug might be linked to issue 700.