Universal-Debloater-Alliance/universal-android-debloater-next-generation #705: feat(scope): Add icons besides app name

agentflemme, 2024-10-29:
### Describe the feature you want

Following #64 as it was closed because of the author's late response, I am asking if it could be possible to add icons to better see what the app is about, with :

 - The general chances of breaking anything if this app is uninstalled as a color
 - What the app does, as an icon ( like a driver, a background service, a configuration app, etc )
 
For the apps that do have an icon, maybe add it in the description or alongside this one ?

So, for example, we could see a system app, say the display driver, should have the color Red (will break the system if removed) and a spanner in it ( for a driver )
Another example, the Google calculator app, should have the color Yellow (will remove functionality, but is not system-breaking and is reversible) with a display in it (as it is an app the user can interact with)

There indeed is a Recommended/Advanced etc categories, but I think it could help users when they go through the list of apps faster when they want to uninstall a custom set of apps they don't need

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-11-23:
> Following #64 as it was closed because of the author's late response

CC @Frigyes06

BTW, thanks for referencing the original issue.

> chances of breaking anything if this app is uninstalled as a color

Color-coded `removal`? Seems like a good idea! I think it should only be done for the "summary" view, not the "detailed" view (the one that shows the pack description). This doesn't necessarily exclude color-blind people, but it feels like they would be "missing out" 🙁

> What the app does, as an icon ( like a driver, a background service, a configuration app, etc )

That would require new metadata to be added to The List, or some "magic" detection based on the APK on the device (hard). I would add types such as "content provider"/"data-base" and "live wallpaper".

> This issue is not a duplicate

Essentially, it is, but you also included other features, so I'm unsure if this should be "merged" back to the original :confused:

AnonymousWP, 2024-11-25:
It's fine as-is. Let's keep this one. For more info, we can click through to the other referenced issue.