Universal-Debloater-Alliance/universal-android-debloater-next-generation #706: com.android.rkpdapp NOT safe to remove

avvalex, 2024-10-30:
### Describe the problem

Hi, i've noticed that uninstalling com.android.rkpdapp the APP IO (Italian public administration app) doesn't work properly (something related with the generation of encryption keys).
Re-installing it solves the problem, so maybe it should not be marked ad "recommended" to uninstall.
Bye.

### Provide logs

.

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-11-10:
Thanks for the report, I'll look into it later.

schbrns, 2024-11-25:
Maybe you shouldn't install apps that require Remote Key Provisioning Daemon for absolutely no reason. 

Why would your Italian public administration app require remote provisioning? 

It's literally an enterprise feature. This isn't breakage, it's poor app design.

Please feel free to re-open the issue if you have any questions. Thank you.

closed 2024-11-25 by schbrns